NodeMCU-Tool/NodeMCU-Tool #2: json format output for commands

soldair, 2016-01-24:
in #1 i mentiond a `--format json` or `--json` flag

i said:
"I do think the tool should support something like a --json or --format json flag so commands like this can be consumed by other programs. This would allow refactor for readability of human output without introducing a breaking change. ill open an issue for discussion."

- when we refactor for readability in the "human" version of the output. Programs parsing the output of nodemcu-tool will break. 
- If we provide an alternative computer friendly output format we don't have to worry about it and this tool becomes quite a bit more useful.

AndiDittrich, 2016-01-25:
it's a great idea! for all operations or only for a special set ?

if you're using Node.js you can directly access the Connector API for low-level operations

AndiDittrich, 2016-03-28:
Dear soldair,

i've added a JSON output option to the `devices` and `fsinfo` command. Additionally a global `--silent` option is available to reject status/logging messages.

i'm currently working on some return_codes to indicate program errors to an external process

soldair, 2016-03-29:
cool. ill check it out.  thanks!

closed 2016-07-22 by AndiDittrich