NodeMCU-Tool/NodeMCU-Tool #6: nodemcu-tool: not command found

MeGaPk, 2016-03-25:
After install on archlinux and osx: 
[megapk@MeGaPk-PC ~]$ nodemcu-tool
bash: nodemcu-tool: command not found

install log (on archlinux):
```
[megapk@MeGaPk-PC ~]$ npm install nodemcu-tool

> serialport@2.0.6 install /home/megapk/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/megapk/node_modules/serialport/build/Release/node-v47-linux-x64/serialport.node" is installed via remote
/home/megapk
└─┬ nodemcu-tool@1.4.0 
  ├── cli-progress@1.1.1 
  ├── colors@1.1.2 
  ├─┬ commander@2.9.0 
  │ └── graceful-readlink@1.0.1 
  ├─┬ prompt@0.2.14 
  │ ├── pkginfo@0.4.0 
  │ ├─┬ read@1.0.7 
  │ │ └── mute-stream@0.0.6 
  │ ├── revalidator@0.1.8 
  │ ├─┬ utile@0.2.1 
  │ │ ├── async@0.2.10 
  │ │ ├── deep-equal@1.0.1 
  │ │ ├── i@0.3.4 
  │ │ ├── mkdirp@0.5.1 
  │ │ ├── ncp@0.4.2 
  │ │ └─┬ rimraf@2.5.2 
  │ │   └─┬ glob@7.0.3 
  │ │     ├─┬ inflight@1.0.4 
  │ │     │ └── wrappy@1.0.1 
  │ │     ├── inherits@2.0.1 
  │ │     ├─┬ minimatch@3.0.0 
  │ │     │ └─┬ brace-expansion@1.1.3 
  │ │     │   ├── balanced-match@0.3.0 
  │ │     │   └── concat-map@0.0.1 
  │ │     ├── once@1.3.3 
  │ │     └── path-is-absolute@1.0.0 
  │ └─┬ winston@0.8.3 
  │   ├── colors@0.6.2 
  │   ├── cycle@1.0.3 
  │   ├── eyes@0.1.8 
  │   ├── isstream@0.1.2 
  │   ├── pkginfo@0.3.1 
  │   └── stack-trace@0.0.9 
  └─┬ serialport@2.0.6 
    ├── async@0.9.0 
    ├── bindings@1.2.1 
    ├─┬ debug@2.2.0 
    │ └── ms@0.7.1 
    ├── nan@2.0.9 
    ├─┬ node-pre-gyp@0.6.18 
    │ ├─┬ mkdirp@0.5.1 
    │ │ └── minimist@0.0.8 
    │ ├─┬ nopt@3.0.6 
    │ │ └── abbrev@1.0.7 
    │ ├─┬ npmlog@2.0.0 
    │ │ ├── ansi@0.3.0 
    │ │ ├─┬ are-we-there-yet@1.0.5 
    │ │ │ ├── delegates@0.1.0 
    │ │ │ └─┬ readable-stream@2.0.5 
    │ │ │   ├── core-util-is@1.0.2 
    │ │ │   ├── inherits@2.0.1 
    │ │ │   ├── isarray@0.0.1 
    │ │ │   ├── process-nextick-args@1.0.6 
    │ │ │   ├── string_decoder@0.10.31 
    │ │ │   └── util-deprecate@1.0.2 
    │ │ └─┬ gauge@1.2.2 
    │ │   ├── has-unicode@1.0.1 
    │ │   ├─┬ lodash.pad@3.1.1 
    │ │   │ ├── lodash._basetostring@3.0.1 
    │ │   │ └─┬ lodash._createpadding@3.6.1 
    │ │   │   └── lodash.repeat@3.0.1 
    │ │   ├─┬ lodash.padleft@3.1.1 
    │ │   │ ├── lodash._basetostring@3.0.1 
    │ │   │ └─┬ lodash._createpadding@3.6.1 
    │ │   │   └── lodash.repeat@3.0.1 
    │ │   └─┬ lodash.padright@3.1.1 
    │ │     ├── lodash._basetostring@3.0.1 
    │ │     └─┬ lodash._createpadding@3.6.1 
    │ │       └── lodash.repeat@3.0.1 
    │ ├─┬ rc@1.1.5 
    │ │ ├── deep-extend@0.4.0 
    │ │ ├── ini@1.3.4 
    │ │ ├── minimist@1.2.0 
    │ │ └── strip-json-comments@1.0.4 
    │ ├─┬ request@2.67.0 
    │ │ ├── aws-sign2@0.6.0 
    │ │ ├─┬ bl@1.0.0 
    │ │ │ └─┬ readable-stream@2.0.5 
    │ │ │   ├── core-util-is@1.0.2 
    │ │ │   ├── inherits@2.0.1 
    │ │ │   ├── isarray@0.0.1 
    │ │ │   ├── process-nextick-args@1.0.6 
    │ │ │   ├── string_decoder@0.10.31 
    │ │ │   └── util-deprecate@1.0.2 
    │ │ ├── caseless@0.11.0 
    │ │ ├─┬ combined-stream@1.0.5 
    │ │ │ └── delayed-stream@1.0.0 
    │ │ ├── extend@3.0.0 
    │ │ ├── forever-agent@0.6.1 
    │ │ ├─┬ form-data@1.0.0-rc3 
    │ │ │ └── async@1.5.0 
    │ │ ├─┬ har-validator@2.0.3 
    │ │ │ ├─┬ chalk@1.1.1 
    │ │ │ │ ├── ansi-styles@2.1.0 
    │ │ │ │ ├── escape-string-regexp@1.0.3 
    │ │ │ │ ├─┬ has-ansi@2.0.0 
    │ │ │ │ │ └── ansi-regex@2.0.0 
    │ │ │ │ ├─┬ strip-ansi@3.0.0 
    │ │ │ │ │ └── ansi-regex@2.0.0 
    │ │ │ │ └── supports-color@2.0.0 
    │ │ │ ├─┬ commander@2.9.0 
    │ │ │ │ └── graceful-readlink@1.0.1 
    │ │ │ ├─┬ is-my-json-valid@2.12.3 
    │ │ │ │ ├── generate-function@2.0.0 
    │ │ │ │ ├─┬ generate-object-property@1.2.0 
    │ │ │ │ │ └── is-property@1.0.2 
    │ │ │ │ ├── jsonpointer@2.0.0 
    │ │ │ │ └── xtend@4.0.1 
    │ │ │ └─┬ pinkie-promise@2.0.0 
    │ │ │   └── pinkie@2.0.1 
    │ │ ├─┬ hawk@3.1.2 
    │ │ │ ├── boom@2.10.1 
    │ │ │ ├── cryptiles@2.0.5 
    │ │ │ ├── hoek@2.16.3 
    │ │ │ └── sntp@1.0.9 
    │ │ ├─┬ http-signature@1.1.0 
    │ │ │ ├── assert-plus@0.1.5 
    │ │ │ ├─┬ jsprim@1.2.2 
    │ │ │ │ ├── extsprintf@1.0.2 
    │ │ │ │ ├── json-schema@0.2.2 
    │ │ │ │ └── verror@1.3.6 
    │ │ │ └─┬ sshpk@1.7.1 
    │ │ │   ├── asn1@0.2.3 
    │ │ │   ├── assert-plus@0.2.0 
    │ │ │   ├─┬ dashdash@1.10.1 
    │ │ │   │ └── assert-plus@0.1.5 
    │ │ │   ├── ecc-jsbn@0.1.1 
    │ │ │   ├── jodid25519@1.0.2 
    │ │ │   ├── jsbn@0.1.0 
    │ │ │   └── tweetnacl@0.13.2 
    │ │ ├── is-typedarray@1.0.0 
    │ │ ├── isstream@0.1.2 
    │ │ ├── json-stringify-safe@5.0.1 
    │ │ ├─┬ mime-types@2.1.8 
    │ │ │ └── mime-db@1.20.0 
    │ │ ├── node-uuid@1.4.7 
    │ │ ├── oauth-sign@0.8.0 
    │ │ ├── qs@5.2.0 
    │ │ ├── stringstream@0.0.5 
    │ │ ├── tough-cookie@2.2.1 
    │ │ └── tunnel-agent@0.4.2 
    │ ├─┬ rimraf@2.4.4 
    │ │ └─┬ glob@5.0.15 
    │ │   ├─┬ inflight@1.0.4 
    │ │   │ └── wrappy@1.0.1 
    │ │   ├── inherits@2.0.1 
    │ │   ├─┬ minimatch@3.0.0 
    │ │   │ └─┬ brace-expansion@1.1.2 
    │ │   │   ├── balanced-match@0.3.0 
    │ │   │   └── concat-map@0.0.1 
    │ │   ├─┬ once@1.3.3 
    │ │   │ └── wrappy@1.0.1 
    │ │   └── path-is-absolute@1.0.0 
    │ ├── semver@5.1.0 
    │ ├─┬ tar@2.2.1 
    │ │ ├── block-stream@0.0.8 
    │ │ ├─┬ fstream@1.0.8 
    │ │ │ └── graceful-fs@4.1.2 
    │ │ └── inherits@2.0.1 
    │ └─┬ tar-pack@3.1.2 
    │   ├── debug@0.7.4 
    │   ├─┬ fstream@1.0.8 
    │   │ ├── graceful-fs@4.1.2 
    │   │ └── inherits@2.0.1 
    │   ├─┬ fstream-ignore@1.0.3 
    │   │ ├── inherits@2.0.1 
    │   │ └─┬ minimatch@3.0.0 
    │   │   └─┬ brace-expansion@1.1.2 
    │   │     ├── balanced-match@0.3.0 
    │   │     └── concat-map@0.0.1 
    │   ├── once@1.1.1 
    │   ├─┬ readable-stream@2.0.5 
    │   │ ├── core-util-is@1.0.2 
    │   │ ├── inherits@2.0.1 
    │   │ ├── isarray@0.0.1 
    │   │ ├── process-nextick-args@1.0.6 
    │   │ ├── string_decoder@0.10.31 
    │   │ └── util-deprecate@1.0.2 
    │   └── uid-number@0.0.3 
    ├─┬ node-pre-gyp-github@1.1.1 
    │ ├─┬ commander@2.9.0 
    │ │ └── graceful-readlink@1.0.1 
    │ ├── github@0.2.4 
    │ ├── mime@1.3.4 
    │ └─┬ request@2.67.0 
    │   ├── aws-sign2@0.6.0 
    │   ├─┬ bl@1.0.0 
    │   │ └─┬ readable-stream@2.0.5 
    │   │   ├── core-util-is@1.0.2 
    │   │   ├── inherits@2.0.1 
    │   │   ├── isarray@0.0.1 
    │   │   ├── process-nextick-args@1.0.6 
    │   │   ├── string_decoder@0.10.31 
    │   │   └── util-deprecate@1.0.2 
    │   ├── caseless@0.11.0 
    │   ├─┬ combined-stream@1.0.5 
    │   │ └── delayed-stream@1.0.0 
    │   ├── extend@3.0.0 
    │   ├── forever-agent@0.6.1 
    │   ├─┬ form-data@1.0.0-rc3 
    │   │ └── async@1.5.0 
    │   ├─┬ har-validator@2.0.3 
    │   │ ├─┬ chalk@1.1.1 
    │   │ │ ├── ansi-styles@2.1.0 
    │   │ │ ├── escape-string-regexp@1.0.4 
    │   │ │ ├─┬ has-ansi@2.0.0 
    │   │ │ │ └── ansi-regex@2.0.0 
    │   │ │ ├─┬ strip-ansi@3.0.0 
    │   │ │ │ └── ansi-regex@2.0.0 
    │   │ │ └── supports-color@2.0.0 
    │   │ ├─┬ is-my-json-valid@2.12.3 
    │   │ │ ├── generate-function@2.0.0 
    │   │ │ ├─┬ generate-object-property@1.2.0 
    │   │ │ │ └── is-property@1.0.2 
    │   │ │ ├── jsonpointer@2.0.0 
    │   │ │ └── xtend@4.0.1 
    │   │ └─┬ pinkie-promise@2.0.0 
    │   │   └── pinkie@2.0.1 
    │   ├─┬ hawk@3.1.2 
    │   │ ├── boom@2.10.1 
    │   │ ├── cryptiles@2.0.5 
    │   │ ├── hoek@2.16.3 
    │   │ └── sntp@1.0.9 
    │   ├─┬ http-signature@1.1.0 
    │   │ ├── assert-plus@0.1.5 
    │   │ ├─┬ jsprim@1.2.2 
    │   │ │ ├── extsprintf@1.0.2 
    │   │ │ ├── json-schema@0.2.2 
    │   │ │ └── verror@1.3.6 
    │   │ └─┬ sshpk@1.7.1 
    │   │   ├── asn1@0.2.3 
    │   │   ├── assert-plus@0.2.0 
    │   │   ├─┬ dashdash@1.10.1 
    │   │   │ └── assert-plus@0.1.5 
    │   │   ├── ecc-jsbn@0.1.1 
    │   │   ├── jodid25519@1.0.2 
    │   │   ├── jsbn@0.1.0 
    │   │   └── tweetnacl@0.13.2 
    │   ├── is-typedarray@1.0.0 
    │   ├── isstream@0.1.2 
    │   ├── json-stringify-safe@5.0.1 
    │   ├─┬ mime-types@2.1.8 
    │   │ └── mime-db@1.20.0 
    │   ├── node-uuid@1.4.7 
    │   ├── oauth-sign@0.8.0 
    │   ├── qs@5.2.0 
    │   ├── stringstream@0.0.5 
    │   ├── tough-cookie@2.2.1 
    │   └── tunnel-agent@0.4.2 
    ├─┬ optimist@0.6.1 
    │ ├── minimist@0.0.8 
    │ └── wordwrap@0.0.3 
    └── sf@0.1.7 

npm WARN enoent ENOENT: no such file or directory, open '/home/megapk/package.json'
npm WARN megapk No description
npm WARN megapk No repository field.
npm WARN megapk No README data
npm WARN megapk No license field.
```

AndiDittrich, 2016-03-25:
Dear MeGaPk,

you have installed **NodeMCU-Tool** as LOCAL package - this means that the binary file ist **not exposed** to your systems path!

in this case, you have to execute the file within the `bin` dir:

```bash
$ ./node_modules/nodemcu-tool/bin/nodemcu-tool --version
1.4.0
```

or install the package in global scope by using the `-g` flag to register the command within your path

```bash
$ npm install nodemcu-tool -g
```

best regards, Andi

MeGaPk, 2016-03-25:
Maybe add in guide? To avoid this problem for other people.

AndiDittrich, 2016-03-25:
of course! i will add it to the next release. thanks for this suggestion

closed 2016-03-25 by AndiDittrich

MeGaPk, 2016-03-25:
And don't forget use "sudo" too :)

```
npm ERR! Error: EACCES: permission denied, access '/usr/lib/node_modules'
npm ERR!     at Error (native)
npm ERR!  { [Error: EACCES: permission denied, access '/usr/lib/node_modules']
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/lib/node_modules' }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/megapk/node_modules/nodemcu-tool/bin/npm-debug.log
```

AndiDittrich, 2016-03-25:
Normally, `sudo` should not be required, because the path `usr/lib/` should be world readable & executable