NodeMCU-Tool/NodeMCU-Tool #7: nodemcu-tool can't open files

curioussavage, 2016-03-26:
Hi, I'm using this on my raspberry pi and keep getting this error. 

```
[NodeMCU-Tool] Connected
[NodeMCU] Version: 0.9.6 | ChipID: 0xd676da | FlashID: 0x1640e0
[NodeMCU] Running "main.lua"
>----------------------------->
cannot open main.lua
>----------------------------->
```
I used `chmod` to change permissions on the file but I still can't run it.

AndiDittrich, 2016-03-26:
Dear curioussavage,

i cannot reproduce the error. the **run** command will execute a lua/lc file on NodeMCU and NOT on the local file systems - this means you cannot change anything with `chmod`

what does the output of `nodemcu-tool fsinfo` show ? does the file exists on your NodeMCU Module ?
do you've tried to restart the module (power off/on) ?

curioussavage, 2016-03-26:
AndiDittrich, thanks taking the time to respond. I misunderstood the description of run, as you indicated. I thought it would send the contents of a lua file to the device to be executed. Tried it with a file that I uploaded and it works fine. 

Thanks again. And nice tool you have made here!

closed 2016-03-26 by curioussavage