NodeMCU-Tool/NodeMCU-Tool #8: Execute arbitrary Lua statements

remcoder, 2016-03-31:
It would be a nice addition to be able to run this for instance:
`node.restart()`

Executing commands like this via nodemcu-tool would allow be to easily write all sorts of bash scripts to support my dev cycle. Or is there already a way to do this?

AndiDittrich, 2016-04-01:
Dear Remco,

such option is currently not available via the CLI but i can implement it. for basic commands like "restart" you can create a file like `restart.lua` with a special set of commands and run it directly:

```bash
$ nodemcu-tool run restart.lua
```

best regards, Andi

remcoder, 2016-04-01:
Hah, of course! :D Thanks for pointing that out.

I guess there's not really a need to implement anything extra then.

remcoder, 2016-04-01:
Thanks for your work on nodemcu-tool btw. It really eases the dev flow with nodemcu. Keep up the good work! ;-)

AndiDittrich, 2016-04-01:
ok & thanks!

closed 2016-04-01 by AndiDittrich