NodeMCU-Tool/NodeMCU-Tool #10: module mismatch

rgujju, 2016-05-21:
i am getting a module mismatch error when i try to use this in my node.js project. i am using node 4.2.6
nodemcu-tool is intalled globally.

A JavaScript error occurred in the main process
Uncaught Exception:
Error: Module version mismatch. Expected 48, got 46.
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:158:20)
    at Object.Module._extensions..node (module.js:568:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:158:20)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at bindings (/home/rgujju/Documents/My Apps/espUploader/node_modules/bindings/bindings.js:76:44)

AndiDittrich, 2016-05-22:
Dear rgujju,

do you have updated your nodejs version to a newer one **without** running `npm update` ?
it is a common nodejs issue.

rgujju, 2016-05-22:
I guess that was the problem. I updated nodejs and ran `npm update` its working perfectly now. 
Thank You!

closed 2016-05-22 by AndiDittrich