NodeMCU-Tool/NodeMCU-Tool #11: keep terminal open after run command

blezek, 2016-05-27:
Would like an option to keep the terminal open after a `run` command.  Because the ESP8266 is event driven, often my scripts produce output long after `NodeMCU-Tool` closes the serial terminal after running a lua file.  A flag to the command would be most helpful.

AndiDittrich, 2016-05-28:
hi blezek,

currently, the `run` command only waits for a incommming "new line" from the ESP8266 - sequence `\n\r>`. it is may possible to redirect every incomming message to stdout, but it requires some deeper modifications of the application structure. 

As a result of such modification, NodeMCU would keep the connection open for a infinite time and has to be killed by an external process...

are you using NodeMCU-Tool within an external application ?

blezek, 2016-05-28:
@AndiDittrich Appreciate you getting back to me.

Here's my typical use case.

1. write Lua code, with `print` statements for debugging
2. run `nodemcu-tool --port /dev/tty.w* --baud 74880 upload src/blynk.lua` to upload to the ESP8266
3. run `nodemcu-tool --port /dev/tty.w* --baud 74880 terminal`
4. type `dofile ( 'blynk.lua' )`
5. watch output to see what happens
6. repeat

If I could have the `terminal` command execute the `dofile`, or have the `run` command keep the terminal open, then my debugging cycle would be a bit shorter.  The `run` is nice, but my debugging statements sometimes take a few seconds to appear (like connecting to a WiFi network) that I couldn't see them.

This make more sense?

AndiDittrich, 2016-05-28:
Hi Daniel,

my suggestion is to add a `--run` flag to the terminal action, which transmits a `dofile` command initially and continues as terminal session - it only requires a few new lines of code - what do you think ?

with this enhancement you could create a simple shell script like this to speed up the development process

**debug.sh**
```bash
#!/bin/bash

# upload the file
nodemcu-tool upload src/blynk.lua

# run the file "blynk.lua" within a new nodemcu terminal session
nodemcu-tool terminal --run "blynk.lua"
```

this shall reduce your workload to a single cli statement like `./debug.sh`

btw. you should consider to create a [project based configuration](https://github.com/AndiDittrich/NodeMCU-Tool#project-based-configuration) to store the tty settings

blezek, 2016-05-28:
Hi @AndiDittrich 

Your suggestion would be fantastic!  I really like your tool, between it and `esptool.py` to flash the firmware, and emacs, I have a great development setup for the ESP chip.

Let me know when it's ready, and I'll test it out.

Thanks,
-dan

AndiDittrich, 2016-05-28:
ready!

**Example 1**
```shell
$ nodemcu-tool terminal --run helloworld.lua
[SerialTerminal] Starting Terminal Mode - press ctrl+c to exit
dofile("helloworld.lua")
Hello World!
|---|
| H |
| E |
| L |
| L |
| O |
| | |
| W |
| O |
| R |
| L |
| D |
|---|
YEAH!!! HELLO WORLD!!!
String: Lorem ipsum dolor sit amet, consetetur sadipscing elitr
```

blezek, 2016-05-29:
Perfect, looking forward to trying it out soon.

closed 2016-05-29 by blezek