NodeMCU-Tool/NodeMCU-Tool #12: --silent is not completely silent

remcoder, 2016-07-19:
I'd like to consume the output of say the fsinfo command with other tools but the `--silent` option doesn't remove all status messages. Here's an example when using the json dump:

	 ~ > projects > q42 > pingpongnotify $ nodemcu-tool --silent fsinfo --json
	[NodeMCU-Tool] Project based configuration loaded
	{"files":[{"name":"init.lua","size":503},{"name":"patterns/clear.lua","size":88},{"name":"mqttclient.lua","size":1236},{"name":"patterns/rotate_green.lua","size":892},{"name":"patterns/rotate_blue.lua","size":746},{"name":"patterns/green.lua","size":105},{"name":"patterns/test.lua","size":470},{"name":"wifi.lua","size":257},{"name":"patterns/red.lua","size":106},{"name":"commands/reboot.lua","size":42},{"name":"patterns/blue.lua","size":123},{"name":"commands/ip.lua","size":24},{"name":"utils.lua","size":4502},{"name":"animator.lua","size":1276},{"name":"main.lua","size":138},{"name":"httpserver.lua","size":2059},{"name":"commands/test.lua","size":59}],"meta":{"remaining":3341,"used":18,"total":3360}}

As a temporary workaround I'm using `tail` to skip the first line:

    nodemcu-tool fsinfo --json | tail -n +2 | ..etc

AndiDittrich, 2016-07-22:
Dear remcoder,
thanks for your report!

i've forgot to add the output suppression to the binary/executable file. therefore some messages will be printed. i will fix it asap.

best regards, Andi

AndiDittrich, 2016-07-24:
Fix is added to 1.7.0-BETA1.

i've modified the overall log-management of NodeMCU-Tool by using [logging-facility](https://www.npmjs.com/package/logging-facility)

status messages are completely suppressed by using the `--silent` or `--json` flag

closed 2016-07-24 by AndiDittrich

remcoder, 2016-07-27:
Nice!

commit 389054ee (2016-07-24): fixed #12 - used logging-facility for global log management