NodeMCU-Tool/NodeMCU-Tool #13: add option to output fsinfo as raw filenames

remcoder, 2016-07-19:
Consuming the fsinfo output with other tools is cumbersome currently. Here's a real life example.

I wanted to remove all *.lc files from the filesystem and I figured I could just iterate over the list of files using a simple bash script. This was a little more involved as:
 - the default output has additional cruft (although it's nice for a human use)
 - json output is actually not that easy to use from a bash script
 - status messages are printed despite `--silent` (see #12)

The script that I ended up with was this:

    for file in `nodemcu-tool fsinfo --json | tail -n +2 | jq .files[].name -r | grep .lc`
    do
      echo "removing ${file}"
      nodemcu-tool remove ${file}
    done

Here I used `tail` to skip the first line, which is a status message. Then I used `jq` to parse the json output and turn it into a plain list of filenames.

I think this should be as easy as:

    for file in `nodemcu-tool fsinfo --raw | grep .lc`
    do
      echo "removing ${file}"
      nodemcu-tool remove ${file}
    done

AndiDittrich, 2016-07-22:
maybe it is a better idea to add wildcard/glob support to the `remove` command ?

adding support for a "raw" format to the `fsinfo` command is pretty easy, if you like i will add it asap

remcoder, 2016-07-22:
I guess wildcard support for `remove` would be the best solution for the problem I had. This probably more directly useful to other users as well.

AndiDittrich, 2016-07-24:
`--raw` option is added for the `fsinfo` command. you can try it out using the current master branch

The Format is: `<filename>\t<size>\n`

**Example 1 - RAW Format**
```
andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool fsinfo --raw
dev/f2.lc       96
helloworld.lua  382
f3.lua  19
f2.lua  19
helloworld.lc   492
dev/f1.lc       96
dev/f3.lc       96
f1.lua  18
```

**Example 2 - Extract the filenames**
```
andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool fsinfo --raw | cut -f1
dev/f2.lc
helloworld.lua
f3.lua
f2.lua
helloworld.lc
dev/f1.lc
dev/f3.lc
f1.lua
```

closed 2016-07-25 by AndiDittrich

AndiDittrich, 2016-07-29:
changed :)

remcoder, 2016-07-29:
That's actually really cool! Awesome ;-)