NodeMCU-Tool/NodeMCU-Tool #14: NodeMCU v1.5.1

pctj101, 2016-07-21:
I'm getting this error message often:

```
[NodeMCU-Tool] Project based configuration loaded
[NodeMCU-Tool] Unable to establish connection - No response detected - is NodeMCU online and the LUA interpreter ready ?
```

Simply re-running the command between 1 - 20 times eventually gets success.

I never had such issues in 0.9.x NodeMCU releases.

I've looked at the source and see that the connection is a simple connect, echo and string compare, so I can't see why it wouldn't work.

Any ideas? :)

pctj101, 2016-07-21:
Oh interesting.  I'm printing the result of the echo command and....
```
[NodeMCU-Tool] Unable to establish connection - No response detected - is NodeMCU online and the LUA interpreter ready ? <sl���

NodeMCU custom build by frightanic.com
	branch: master
	commit: b580bfe79e6e73020c2bd7cd92a6afe01a8bc867
	SSL: true
	modules: adc,apa102,bit,cjson,coap,crypto,dht,encoder,enduser_setup,file,gpio,http,i2c,mdns,mqtt,net,node,perf,pwm,spi,struct,tmr,uart,wifi,ws2812
 build 	built on: 2016-07-19 13:49
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
lua: cannot open init.lua>
```
I think my ESP is resetting.  Not sure why.

pctj101, 2016-07-21:
Okay I'm guessing that the WeMOS D1 resets on a new connection for convenience.

Problem #1 
The print() command gets sent during boot and is entirely ignored
Delaying print helps, but... then problem #2

Problem #2 
The boot process outputs garbage + info... of which the "ending part" is included as part of the print() request and totally screws up the echo check (even if print() succeeds)

pctj101, 2016-07-21:
Okay PR #15 works for me so far, but I'm not sure how that would affect older devices.  Maybe you have some insight?

AndiDittrich, 2016-07-22:
Dear pctj101,

i have reject your pull request, because you are using synchronous code which may cause an unexcpected behaviour in some cases: the serial-io is async, therefore it can cause random delays between the command and its execution. in some cases the two **echo** commands can overlap and this will cause a fatal error.

currently, i am not shure whats going on in NodeMCU v1.5.X - a reset should not happen during the transfer. i will investigate this issue - maybe they have changed the API again and one of the commands used by NodeMCU-Tool cause the reset...

best regards, Andi

pctj101, 2016-07-24:
Thanks for the detailed rejection! :P

I'll take a peek later if I get a chance to see if there's a way i can improve it.

AndiDittrich, 2016-07-25:
Dear pctj101,

i've just updated my NodeMCU Dev Kit to v1.5.1 (cjson, file, gpio, mqtt, net, node, tmr, uart, wifi) using http://nodemcu-build.com/ and i cannot reproduce the problems you've mentioned.

which build options do you use ? do you have re-created the file-system after flashing the firmware ?

**Step 1 - Firmware Flash**
```
andi@sapphire:~/NodeMCU-Tool$ ./esptool .firmware/nodemcu-master-9-modules-2016-07-25-12-16-08-float.bin
esptool.py v1.1
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 409600 @ 0x0... 409600 (100 %)
Wrote 409600 bytes at 0x0 in 35.5 seconds (92.3 kbit/s)...
Leaving...
```

**Step 2 - Power on/off the Module**

**Step 3 - Re-Create the File System**
```
andi@sapphire:~/NodeMCU-Tool$ ./nodemcu-tool mkfs
[NodeMCU-Tool] Project based configuration loaded
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool] Connected
[NodeMCU] Version: 1.5.1 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU] Formatting the file system...this will take around ~30s
[NodeMCU] File System created | format done.
```

**Step 4 - Upload + Run LUA Files**
```
andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool upload helloworld.lua
[NodeMCU-Tool] Project based configuration loaded
[NodeMCU-Tool] Connected
[NodeMCU] Version: 1.5.1 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU-Tool] Uploading "helloworld.lua" >> "helloworld.lua"...
[NodeMCU-Tool] File Transfer complete!

andi@sapphire:~/NodeMCU-Tool$ ./nodemcu-tool run helloworld.lua
[NodeMCU-Tool] Project based configuration loaded
[NodeMCU-Tool] Connected
[NodeMCU] Version: 1.5.1 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU] Running "helloworld.lua"
[NodeMCU] >----------------------------->
Hello World!
|---|
| H |
| E |
| L |
| L |
| O |
| | |
| W |
| O |
| R |
| L |
| D |
|---|
YEAH!!! HELLO WORLD!!!
String: Lorem ipsum dolor sit amet, consetetur sadipscing elitr
[NodeMCU] >----------------------------->
```

pctj101, 2016-07-26:
Hi Andi,

FYI the modules I included are:
```
This was built against the master branch and includes the following modules: adc, apa102, bit, cjson, coap, crypto, dht, encoder, enduser_setup, file, gpio, http, i2c, mdns, mqtt, net, node, perf, pwm, spi, struct, tmr, uart, wifi, ws2812.
```

I'm using a Wemos D1 which may for whatever reason have DTR/RTS hooked up in some way that makes the board reset "on connect" which it certainly seems to love doing (reset nodemcu when I "open the serial port on my mac")

I did 'mkfs' after flashing my device and file uploads work "great" when they work.

In your case if your board does not reset, then the code I provided is of course not necessary.

The code I provided effectively skips over the boot messages and only looks for the echo response at the very end of whatever comes back over the serial line.

I have other boards that don't auto-reboot because that's "not how it's wired" and will probably have no issue.

Do you happen to have any chronic-rebooting boards? :)

pctj101, 2016-07-26:
For fun, I'm trying to build an image (master) that follows your selected modules.  Will give you an update when I get the output and give it a spin. :)

pctj101, 2016-07-26:
Yeah so on a custom board I have that does not have the luxury of DTR/RTS "software induced" reset, I have no issues at all and life is great.

It's those fancy boards... much in the same way those "Arduinos" have something more fancy than just the "plain atmega chip".

pctj101, 2016-07-26:
Verified the custom non-fancy board does no reset using:
 print(tmr.time())

Both before/after uploading a file.  Uptime only goes up! :)

Almost surely that fancy serial DTR/RTS based reset.

AndiDittrich, 2016-07-26:
Do you have this board http://www.wemos.cc/Products/d1_r2.html ?

The reset circuit on it is taken from the official [NodeMCU DevKit](https://github.com/nodemcu/nodemcu-devkit-v1.0) - i have one of these boards here to test NodeMCU Tool and everything works fine including the rts/dtr hard-reset - therefore, the behaviour of your board is very very strange!

maybe it is a issue with the serial driver which cause the reset.

what happens if you use the terminal mode ? 
```
$ nodemcu-tool terminal
```

does a reset happen every time you use this command ?

pctj101, 2016-07-27:
This is the board I have:
http://www.wemos.cc/Products/d1_mini.html

Will check terminal in a bit. From what I remember the board does reset upon a new connection.

pctj101, 2016-07-27:
```

pctj101, 2016-07-27:
So yes! terminal does reset my device

$ nodemcu-tool  -p /dev/tty.whatver terminal
```
[NodeMCU-Tool] Project based configuration loaded
[SerialTerminal] Starting Terminal Mode - press ctrl+c to exit
print(tmr.time()){dd??|?d?|

                           ?
                            d?c<???{?c?c??gg?doo???
                                                   #p??l{d{lp?o?
                                                                ?l

                                                                  c
                                                                   g?<????
                                                                          b??og?l??$`?nn?dns???g

                ??lx?g?
                       r?????

                             #o?|????c??g'??l`?nol`n{???g
                                                         ?l`x?'?

                                                                r?????

                                                                      ??bo?|

                                                                            b??og?
  d`?gn$`'s???o

               ?d`r??n
                      ?l don't use rtc mem data
rl???

NodeMCU custom build by frightanic.com
	branch: master
	commit: b580bfe79e6e73020c2bd7cd92a6afe01a8bc867
	SSL: true
	modules: adc,apa102,bit,cjson,coap,crypto,dht,encoder,enduser_setup,file,gpio,http,i2c,mdns,mqtt,net,node,perf,pwm,spi,struct,tmr,uart,wifi,ws2812
 build 	built on: 2016-07-19 13:49
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
lua: cannot open init.lua
> 
> print(tmr.time())
1
> [SerialTerminal] Connection closed
```

pctj101, 2016-07-27:
Flashed with 
```
This was built against the master branch and includes the following modules: cjson, file, gpio, mqtt, net, node, tmr, uart, wifi.
```
> print(tmr.time())
14
> [SerialTerminal] Connection closed
```
Exit then try again
```
$ nodemcu-tool  -p /dev/tty.whatever terminal
[SerialTerminal] Starting Terminal Mode - press ctrl+c to exit
?D?I?,R?Y?ixYH??@z??l???`(?6???? <- typical reboot garbage
                                ?W????B?3?
                                          ?j芈??????
> 
> print(tmr.time())
10 <-- smaller # indicating reboot
> 
```

pctj101, 2016-07-27:
In case you need it: http://nodemcu-build.com/builds/nodemcu-master-9-modules-2016-07-26-17-49-13-integer.bin

AndiDittrich, 2016-07-29:
Dear pctj101,

i've added the `--connection-delay` option which delays the connection check after opening the serial connection.

please try to upload a file or run a file on your "weak" module as follows:

```
$ ./nodemcu-tool run helloworld.lua --connection-delay 5000 --baud 115200
```

this will delay the connection check by 5000+3000ms 

i've tested this with my NodeMCU DEVKIT by running a hard reset before the command:

```
$ ./nodemcu-tool reset && ./nodemcu-tool run helloworld.lua --connection-delay 5000 --baud 9600
```

pctj101, 2016-07-31:
cool will try

pctj101, 2016-07-31:
Delay works well.

Noticed some hard coded delays of 1000 (ms?) in code.  Could make that a variable as you see fit so that a 2 second total delay isn't hard coded.

NOTE: Tried to play with code
```
// delay the connection process ? may fix issues related to rebooting modules
            if (connectDelay && connectDelay > 1){
                setTimeout(function(){
    
                    // send dummy sequence
                    this.device.write('\n\n\n'  + _luaCommandBuilder.command.echo + '\n\n\n' , function(){

                        // wait 1/3 to get the dummy sequence processed
                        setTimeout(function(){

                            // send dummy sequence
                            this.device.write('\n\n\n'  + _luaCommandBuilder.command.echo + '\n\n\n' , function(){

                                // wait 1/3 to get the dummy sequence processed
                                setTimeout(function(){

                                    // purge received data
                                    this.device.purge(function(){
                                        // connection startup
                                        checkConnection();
                                    }.bind(this));

                                }.bind(this), 300); // Lowered delay #3

                            }.bind(this));

                        }.bind(this), 300); // Lowered delay #2

                    }.bind(this));

                // 2/3 timeout
                }.bind(this), connectDelay); // --connection-delay 300 (Delay #1)
            }else{
                checkConnection();
            }
```

I did notice that if I lowered Delay #2 and #3 below say 200ms, the tool would "lock up" at:
```
[NodeMCU-Tool] Project based configuration loaded
```
But I have no idea why.

Also I noticed that --port <port> doesn't seem to work for whatever reason (again, don't know why but changing defaults directly in source works fine)

AndiDittrich, 2016-07-31:
thanks for your support! the `--port` options is now working again - by the last refactoring, i've used a wrong object key/variable name.

im not sure if it is useful to make the other delays accessable by end-users. they are required for the auto-baud detection of NodeMCU (the first few echo characters have to be purged). maybe 500ms each will be a good value?

pctj101, 2016-08-09:
Hey Andi, 

Sorry a bit tied up and wasn't able to test the "reduced sleep" for autobaud. 

My logs show errors only when reducing below 200ms, so I'm guessing that in lieu of additional information, that 500ms should be plenty enough.

AndiDittrich, 2016-08-13:
Great! Then i will release the v2.0.0 within the next days

closed 2016-08-13 by AndiDittrich

tanviranade, 2017-09-13:
Hi Andi ,
I am a beginner to NodeMCU and not able to understand the discussion on issue regarding unable to establish the connection. I am using version 2.1.1 and trying to connect nodemcu to run simple test file.please help

C:\Users\TANVI>nodemcu-tool --port COM6 --baud=9600 run test.lua
[NodeMCU-Tool] Unable to establish connection - No response detected - is NodeMCU online and the Lua interpreter ready ?

C:\Users\TANVI>nodemcu-tool run helloworld.lua --connection-delay 5000 --baud 115200 --port COM6
[NodeMCU-Tool] Unable to establish connection - Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?

AndiDittrich, 2017-09-14:
@tanviranade please open a new issue!