NodeMCU-Tool/NodeMCU-Tool #16: Does not work with NodeMCU v3

tim-peters, 2016-08-12:
Even if the v3 (LoLin) of NodeMCU seems to be almost identical with v2 (just another Serial-to-USB Chip) this tool seems not to be working with NodeMCU v3. 

``nodemcu-tool devices`` always shows a total of 0 connected devices.
All drivers are set up correctly (I can upload Script e.g. with the Arduino IDE). I've tested this on Windows 10 as well as on Ubuntu. Both with nodemcu-tool installed global.

AndiDittrich, 2016-08-12:
did you try the `--all` option ?

```bash
$ nodemcu-tool devices --all
```

normally, the devices are filtered by their usb pid/vid - in case another chip is used the id's will not match

tim-peters, 2016-08-13:
with the ``--all`` flag set I've found a device at ttyUSB0. Unfortunately I wasn't able to create the initial file system afterwards (the ``connected`` message did not appear, even after minutes of waiting after confirming that I want to format). I'm not sure, what this is caused by and will check back again later.

Thanks for the quick response so far!

AndiDittrich, 2016-08-13:
could you tell the usb vid/pid of your module ? then i can add them to the filter function.

did you tried the latest beta version of NodeMCU Tool ? maybe it is an issue similar to #14 - you can try the following command:

```bash
./nodemcu-tool fsinfo --connection-delay 400 --baud 115200
```

tim-peters, 2016-08-16:
it's ``1a86, pci-1a86_USB2.0-Serial-if00-port0``

With this command I get an error telling me, that ``--connection-delay`` is not an known option.

I'm using version 1.6.0. Where would I get the beta?

AndiDittrich, 2016-08-16:
thanks! i will add it to the module list.
you can use the github master branch to get a copy of the latest beta release

AndiDittrich, 2016-08-22:
well, `0x1a86` is already in the device list ` NodeMCU v1.0 - CH341 Adapter | 0x1a86  QinHeng Electronics` - it should appear in the deviceinfo list

closed 2016-11-12 by AndiDittrich