NodeMCU-Tool/NodeMCU-Tool #18: port is not defined

otbe, 2016-11-12:
Hi,

I installed the latest version of nodemcu-tool, but I get ```ReferenceError: port is not defined``` even with ```--port```.

```
$ nodemcu-tool devices --port=/dev/tty.wchusbserial1410

ReferenceError: port is not defined
    at Object.Tool.devices (/Users/boo/.nvm/versions/node/v6.9.1/lib/node_modules/nodemcu-tool/lib/NodeMCU-Tool.js:408:43)
    at Command.<anonymous> (/Users/boo/.nvm/versions/node/v6.9.1/lib/node_modules/nodemcu-tool/bin/nodemcu-tool.js:373:22)
    at Command.listener (/Users/boo/.nvm/versions/node/v6.9.1/lib/node_modules/nodemcu-tool/node_modules/commander/index.js:301:8)
    at emitTwo (events.js:106:13)
    at Command.emit (events.js:191:7)
    at Command.parseArgs (/Users/boo/.nvm/versions/node/v6.9.1/lib/node_modules/nodemcu-tool/node_modules/commander/index.js:615:12)
    at Command.parse (/Users/boo/.nvm/versions/node/v6.9.1/lib/node_modules/nodemcu-tool/node_modules/commander/index.js:458:21)
    at Object.<anonymous> (/Users/boo/.nvm/versions/node/v6.9.1/lib/node_modules/nodemcu-tool/bin/nodemcu-tool.js:406:6)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
```

nodemcu-tool version is 2.0.0

Whats wrong with my attempt? :)

Thanks!

AndiDittrich, 2016-11-12:
i will check this today

AndiDittrich, 2016-11-12:
Thanks Benjamin!

it was caused by a refactoring issue fixed in **v2.0.1**

closed 2016-11-12 by AndiDittrich

otbe, 2016-11-12:
Works great. Thanks!

commit b47dba7f (2016-11-12): fixed #18 - devices command was broken since the last release (refactoring issue)