NodeMCU-Tool/NodeMCU-Tool #21: Command line option should have higher priority then .nodemcutool file

tae-jun, 2016-12-03:
Hi dear developers

First of all, thanks a lot for the cool tool! I use it really well. I love it!

So, my suggestion is making command line option have higher priority then options in `.nodemcutool` file. Which means, if I run command below:

```sh
nodemcu-tool upload -o -c src/*.lua
```

and if I have `.nodemcutool` file:
```json
{
    "baudrate": "115200",
    "port": "/dev/cu.wchusbserial1420",
    "optimize": false,
    "compile": false,
    "keeppath": false
}
```
It doesn't compile or optimize sources because of `optimize` and `compile` option in `.nodemcutool`. But I think command line options are first!

It would be really grateful if you accept my suggestion 😄 

Thanks

AndiDittrich, 2016-12-03:
Dear Jun,

thats a great idea! 
because of the incomaptibility, such a modification should only take place in a major release. therefore i will project it for **v3**

best regards, Andi

tae-jun, 2016-12-03:
Cool! Thanks for your prompt response 😄 

I look forward to it 👍

AndiDittrich, 2018-02-17:
implemented in **v3**

closed 2018-02-17 by AndiDittrich