NodeMCU-Tool/NodeMCU-Tool #22: Cannot open remote file for write

curioussavage, 2016-12-10:
Hi thanks a million for this tool. I use it exclusively for uploading files to my wemos d1 minis.

I just flashed new firmware on my device with esptool and then use nodemcu-tool mkfs

I suspect this might be a symptom of an issue with that, either way it might be nice to add more detail/ suggestions to the error message if relevant. 

and of course if you have any idea how I can fix this much a appreciated
[image]

here is the output.

curioussavage, 2016-12-10:
As you should be able to see in the screen shot it creates the files on the device but writes nothing to them.

curioussavage, 2016-12-10:
I tested in the terminal and was able to open and write to the empty file created by nodemcu-tool upload so it looks like it might be an issue with the lib and maybe the newest firmware?

AndiDittrich, 2016-12-10:
Strange... i didn't test v1.5.4 yet
did you tried an earlier firmware version ? e.g. 1.5.1 ?

generally, there are no more details available... the fs.write function will only return false..

AndiDittrich, 2016-12-10:
i can confirm the issue with NodeMCU Firmware **v1.5.4** - i will investigate the issue asap

AndiDittrich, 2016-12-10:
should be fixed within the current master branch of NodeMCU-Tool
The team of NodeMCU LUA has changed the footprint of `file.open` within the last minor release...

Thank you very much for your report @curioussavage

curioussavage, 2016-12-10:
That was fast! thanks for all your work on this.

curioussavage, 2016-12-10:
I just successfully tested it on my machine.

closed 2016-12-10 by curioussavage