NodeMCU-Tool/NodeMCU-Tool #23: TypeError: process.stdin.setRawMode is not a function

stephanMettler, 2016-12-17:
Hi,

I was trying to run this awesome tool from a bash inotifywait script.

and got:
```
nodemcu-tool/lib/SerialTerminal.js:53
            process.stdin.setRawMode(true);
                          ^

TypeError: process.stdin.setRawMode is not a function
```

I added this in line ~53

```
 if(process.stdin.isTTY) {
            process.stdin.setRawMode(true);
 }
```

AndiDittrich, 2016-12-17:
Dear stephan,

thanks for your report. i will fix it asap.

btw. what is your use case for the `terminal` command in combination with a bash script ? i've never intended to use the terminal mode in this way ;)

best regards, Andi

closed 2016-12-17 by AndiDittrich

stephanMettler, 2017-01-17:
thank you!
i used it with a "file-changed"-watcher to auto upload after saving.