NodeMCU-Tool/NodeMCU-Tool #25: Unable to connect properly to NodeMCU custom build powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

mohnen, 2017-01-30:
I am running your tool on a Mac. 
`nodemcu-tool devices` gives me this
```
[NodeMCU] Connected Devices | Total: 1
	  |- /dev/cu.wchusbserial1420 (undefined, undefined)
```
Any subsequent command fails, e.g. `nodemcu-tool --port /dev/cu.wchusbserial1420 --baud  115200 mkfs`
```
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) y
[NodeMCU-Tool] Unable to establish connection - No response detected - is NodeMCU online and the Lua interpreter ready ?
```

However, I can connect to the terminal `nodemcu-tool --port /dev/cu.wchusbserial1420 terminal`
```
[SerialTerminal] Starting Terminal Mode - press ctrl+c to exit
```

This is the boot log
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 81ec3665cb5fe68eb8596612485cc206b65659c9
	SSL: false
	modules: adc,cjson,encoder,file,gpio,http,mdns,mqtt,net,node,rtcfifo,rtcmem,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2017-01-12 05:51
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
```

mohnen, 2017-01-30:
I got it fixed with `--connection-delay 50`

AndiDittrich, 2017-02-02:
Great!
Just for interest: what kind of module (vendor/type) do you use ?

mohnen, 2017-02-02:
I used it with two boards, in both cases I needed `--connection-delay 50`
- [NodeMcu Lua ESP-12E WIFI Development Board](https://www.amazon.de/gp/product/B018E741G4/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1)
- [Sonoff](https://www.amazon.de/gp/product/B01M6ZQU6T/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1)

closed 2017-02-02 by AndiDittrich

wolfg1969, 2017-06-28:
I use --connection-delay 5000.

raykin, 2017-10-14:
same problems here.
But the conection-delay option does not work here.
```sh
~/Downloads ᐅ nodemcu-tool mkfs -p /dev/cu.SLAB_USBtoUART --connection-delay 1000 --baud 115200
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool] Unable to establish connection - Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
```
It shows the same error no matter what baud/connection-deplay was set.

raykin, 2017-10-14:
Just reread the doc of https://github.com/nodemcu/nodemcu-devkit-v1.0 and reflash the firmware in  DIO mode and now it works.