NodeMCU-Tool/NodeMCU-Tool #26: Add Support for NodeMCU LUA 2.0.0

ersutton, 2017-02-03:
https://github.com/nodemcu/nodemcu-firmware/releases

Tried to install the latest build from cloud build service here: https://nodemcu-build.com/
and your very useful tool spat the dummy. Any chance of an update to fix?

AndiDittrich, 2017-02-03:
I will check it - regarding to the [Changelog](https://github.com/nodemcu/nodemcu-firmware/milestone/4?closed=1) there are no major changes within this release which breaks nodemcu-tool.

what kind of error did you received ?

AndiDittrich, 2017-02-05:
Dear ersutton,

i cannot confirm the issue - seems to **work without any issues**

**Terminal Session**
```console
andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./esptool .firmware/nodemcu-master-12-modules-2017-02-05-15-06-51-integer.bin
esptool.py v1.1
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 593920 @ 0x0... 593920 (100 %)
Wrote 593920 bytes at 0x0 in 51.5 seconds (92.3 kbit/s)...
Leaving...

andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool reset
[NodeMCU-Tool] Connected
[NodeMCU] Hard-Reset executed (100ms)

andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool fsinfo
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU] Free Disk Space: 3184 KB | Total: 3184 KB | 0 Files
[NodeMCU] No Files found - have you created the file-system?

andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool mkfs
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU] Formatting the file system...this will take around ~30s
[NodeMCU] File System created | format done.

andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool upload helloworld.lua
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU-Tool] Uploading "helloworld.lua" >> "helloworld.lua"...
[NodeMCU-Tool] File Transfer complete!

andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool run helloworld.lua
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU] Running "helloworld.lua"
[NodeMCU] >----------------------------->
Hello World!
|---|
| H |
| E |
| L |
| L |
| O |
| | |
| W |
| O |
| R |
| L |
| D |
|---|
YEAH!!! HELLO WORLD!!!
String: Lorem ipsum dolor sit amet, consetetur sadipscing elitr
[NodeMCU] >----------------------------->
```

ersutton, 2017-02-07:
Sorry for wasting your time, after upgrading to the latest release our batch file works flawlessly again.

closed 2017-02-07 by ersutton