NodeMCU-Tool/NodeMCU-Tool #35: How to upload to a Witty Cloud developer board?

T-vK, 2017-04-27:
I have a couple of these [Witty Cloud development boards](https://www.aliexpress.com/item/10PCS-LOT-ESP8266-serial-WIFI-model-ESP-14-Authenticity-Guaranteed/32592556329.html).

Flashing the NodeMCU firmware works just fine without any problems using esptool.py. And I was also able to use ESplorer to upload my init.lua file.

But I cannot get it to work with NodeMCU-Tool. I tried at 9600 and 115200 baud rates. 
I always get:

> Unable to establish connection - Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?

And I should mention: When using ESplorer, I have to do a short RST followed by a short DTR in the UI after opening the COM port and I have to use a baud rate of 115200. 

Is there maybe some sort of special setting that I need to use in order to make NodeMCU-Tool send an RST and DTR signal?

I'm on Linux Mint x64.

AndiDittrich, 2017-04-27:
Did you tried the `--connection-delay` option ? see https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/Reset_on_Connect.md

T-vK, 2017-04-27:
Awesome! Thank you so much for the quick response. It works :+1:

closed 2017-04-27 by AndiDittrich