NodeMCU-Tool/NodeMCU-Tool #36: Uploading blank files results in error

T-vK, 2017-04-29:
When trying to upload an empty file, I get this error:

> node_modules/nodemcu-tool/lib/NodeMcuConnector.js:240                                                                                      
>                 if (chunks.length > 0){                                                                                                                                                                                 
>                           ^                                                                                                                                                                                             
>                                                                                                                                                                                                                         
> TypeError: Cannot read property 'length' of null

Steps to reproduce on Linux:
``` bash
touch empty.lua
nodemcu-tool upload empty.lua
```

Full output:
```
user@linux /tmp/test $ touch empty.lua                                                                                                                                                                     
user@linux /tmp/test $ nodemcu-tool upload empty.lua
[NodeMCU-Tool] Project based configuration loaded                                                                                                                                                                       
[NodeMCU-Tool] Connected                                                                                                                                                                                                
[NodeMCU] Version: 2.0.0 | ChipID: 0x1676b1 | FlashID: 0x1640e0                                                                                                                                                         
[NodeMCU-Tool] Uploading "empty.lua" >> "empty.lua"...                                                                                                                                                                  
[NodeMCU-Connector] Transfer-Mode: hex                                                                                                                                                                                  
 ░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 1% | ETA: undefineds | 1/100/usr/local/lib/node_modules/nodemcu-tool/lib/NodeMcuConnector.js:240                                                                              
                if (chunks.length > 0){                                                                                                                                                                                 
                          ^                                                                                                                                                                                             
                                                                                                                                                                                                                        
TypeError: Cannot read property 'length' of null                                                                                                                                                                        
    at NodeMcuConnector.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/lib/NodeMcuConnector.js:240:27)                                                                                                           
    at NodeMcuConnector.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/lib/NodeMcuConnector.js:275:13)                                                                                                           
    at ScriptableSerialTerminal.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/lib/ScriptableSerialTerminal.js:122:17)                                                                                           
    at ScriptableSerialTerminal.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/lib/ScriptableSerialTerminal.js:73:15)                                                                                            
    at emitOne (events.js:77:13)                                                                                                                                                                                        
    at SerialPort.emit (events.js:169:7)                                                                                                                                                                                
    at SerialPort.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/node_modules/serialport/lib/parsers.js:57:19)                                                                                                   
    at SerialPort._emitData (/usr/local/lib/node_modules/nodemcu-tool/node_modules/serialport/lib/serialport.js:313:18)                                                                                                 
    at SerialPort.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/node_modules/serialport/lib/serialport.js:293:14)                                                                                               
    at SerialPort.<anonymous> (/usr/local/lib/node_modules/nodemcu-tool/node_modules/serialport/lib/serialport.js:306:7) 
```

AndiDittrich, 2017-04-29:
Don't upload empty files :stuck_out_tongue: 

Seriously, thanks for your report! i will fix it soon.

best regards, Andi

AndiDittrich, 2017-04-29:
fixed in **v2.1.1** - see #37

closed 2017-04-29 by AndiDittrich