NodeMCU-Tool/NodeMCU-Tool #40: `nodemcu-tool --optimize` doesn't work

T-vK, 2017-05-08:
`nodemcu-tool --optimize upload init.lua` results in `[NodeMCU-Tool] Unknown command "init.lua"`. 
Can anyone confirm this?

AndiDittrich, 2017-05-08:
Dear T-vK,

i can confirm this issue...it seems to be an error within [commanderjs](https://github.com/tj/commander.js/) argument parsing. but maybe it's an issue by design because NodeMCU-Tool is using a sub-command structure and the argument options are only added to the specific sub-commands. when commander tries to dispatch a command like **upload** it has to parse the global options first. unknown arguments may parsed in a generic way...

generally the tool should be called like this: 

`nodemcu-tool [global-options] <command> [command-options] args..`

**this causes an error**
it seems that the argument parser uses **upload** as value of `--optimize`
```
andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool -n init.lua --optimize upload init.lua
[NodeMCU-Tool] Unknown command "init.lua"

  Usage: nodemcu-tool [options] [command]
```

**this works..**
maybe because there is an argument required by the n options
```
andi@sapphire:~/Development/IoT/NodeMCU-Tool$ ./nodemcu-tool --optimize -n init.lua upload init.lua
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU-Tool] Uploading "init.lua" >> "init.lua"...
[NodeMCU-Connector] Transfer-Mode: base64
[NodeMCU-Tool] File Transfer complete!
```

T-vK, 2017-05-08:
Wouldn't it make more sense to have `--optimize`, `--keeppath` etc on the right of of `upload`? I mean do you really consider those to be `[global-options]`? I thought they were specific to `upload` and thus should fall under `[command-options]`.

AndiDittrich, 2017-05-08:
i've never said that this are global options, the examples above are only used as testcase ;)

T-vK, 2017-05-08:
Isn't the [CommandReference](https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/CommandReference.md#upload-files) kind of misleading then?

AndiDittrich, 2017-05-08:
of course, i'm currently fixing this

closed 2017-05-08 by AndiDittrich