NodeMCU-Tool/NodeMCU-Tool #43: Not able to run the nodemcu-tool.js

Ankitr19, 2017-07-25:
[image]
Please help!!

AndiDittrich, 2017-07-25:
Hi Ankit,

this seems to be an issue with the [serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport) package - not NodeMCU-Tool.

Please take a look onto their FAQs - in most cases such errors a caused by a missconfigured environment.
did you try to install the tool in the local folder instead of global scope ?

br, Andi

Ankitr19, 2017-07-25:
Sir, now when I am trying to install it locally, i.e. just by cloning
the git repo and running the nodemcu-tool.js it is giving another
error.

AndiDittrich, 2017-07-25:
in this case you have to run `npm install` to fetch the dependencies...

normally a simple `npm install nodemcu-tool` should do the job

Ankitr19, 2017-07-25:
Sir a little query,  when I run npm isntall nodemcu-tool where it gets
install.  I mean where I can find the. js file to execute it?

Ankit Raj,
B. Tech,  Electronics and Communication  Engineering
NIT Agartala
8414846511

AndiDittrich, 2017-07-25:
Please read the readme file first ;)

**in case you've cloned the github repo**
the "executable" file is located in `./bin/nodemcu-tool.js`
https://github.com/AndiDittrich/NodeMCU-Tool#as-archive-from-github

**in case of local npm installation**
the "executable" file is located in `./node_modules/nodemcu-tool/bin/nodemcu-tool.js`
https://github.com/AndiDittrich/NodeMCU-Tool#localproject-related-installation

Ankitr19, 2017-07-25:
Yes, Sir, it worked. :)

closed 2017-07-25 by AndiDittrich