NodeMCU-Tool/NodeMCU-Tool #44: Upload option -k not working

zunware, 2017-09-16:
I tried to upload my files using `nodemcu-tool upload -k kit/*.lua` expecting them to be uploaded to the same relative path (`/kit/*.lua`) but it is copying to the root of FS. 

[image]

AndiDittrich, 2017-09-17:
Hi zunware,
which NodeMCU-Tool version did you use ? does the "full" option "--keeppath" work ?

best regards, Andi

AndiDittrich, 2017-09-17:
Hi zunware,

I've tested your issue but cannot reproduce the behaviour:

```terminal
 $ ./nodemcu-tool --version
2.2.0

 $ ./nodemcu-tool upload -k kit/*.lua
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU-Tool] Uploading "kit/l1.lua" >> "kit/l1.lua"...
[NodeMCU-Connector] Transfer-Mode: base64
[NodeMCU-Tool] Uploading "kit/l2.lua" >> "kit/l2.lua"...
[NodeMCU-Tool] Uploading "kit/l3.lua" >> "kit/l3.lua"...
[NodeMCU-Tool] Uploading "kit/l4.lua" >> "kit/l4.lua"...
[NodeMCU-Tool] Uploading "kit/l5.lua" >> "kit/l5.lua"...
[NodeMCU-Tool] Uploading "kit/l6.lua" >> "kit/l6.lua"...
[NodeMCU-Tool] Uploading "kit/l7.lua" >> "kit/l7.lua"...
[NodeMCU-Tool] Uploading "kit/l8.lua" >> "kit/l8.lua"...
[NodeMCU-Tool] Bulk File Transfer complete!

 $ ./nodemcu-tool fsinfo
[NodeMCU-Tool] Connected
[NodeMCU] Version: 2.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0
[NodeMCU] Free Disk Space: 3152 KB | Total: 3366 KB | 28 Files
[NodeMCU] Files stored into Flash (SPIFFS)
          |- kit/l4.lua (13313 Bytes)
          |- kit/l8.lua (13313 Bytes)
          |- kit/l7.lua (13313 Bytes)
          |- kit/l6.lua (13313 Bytes)
          |- kit/l5.lua (13313 Bytes)
          |- kit/l3.lua (13313 Bytes)
          |- kit/l2.lua (13313 Bytes)
          |- kit/l1.lua (13313 Bytes)
          |- test/sysinfo.lua (741 Bytes)
```

maybe you've set `keeppath=false` in your project configuration ? this will override the cli flag

closed 2017-09-24 by AndiDittrich

zunware, 2017-09-25:
@AndiDittrich you where right. The setting `keeppath=false` was set in my config file. Changing this did the trick!