NodeMCU-Tool/NodeMCU-Tool #45: `nodemcu-tool devices` does not list my device.

KonradKlimczak, 2017-09-25:
Hi,

I just installed `nodemcu-tool` and I wanted to connect to my device. Before that I have used [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher/blob/master/Win64/Release/ESP8266Flasher.exe) from the official repo.

I have "ESP8266 NODEmcu V3".
I am working on Windows 10 Home.
Installed version of nodemcu-tool: 2.1.1

When I try to list my devices I get this:
```
C:\Users\Konrad>nodemcu-tool devices
[NodeMCU] No Connected Devices found | Total: 0
```
What's the problem?

AndiDittrich, 2017-09-25:
Hi Konrad,

please try the `--all` flag. depending on your NodeMCU board/vendor the usb pid/vid may not match the "official" ones

btw. NodeMCU-Tool is not an replacement for esptool or nodemcu-flasher to flash the firmware. it is designed to managed the LUA application files

KonradKlimczak, 2017-09-25:
I am still missing something:

```
C:\Users\Konrad>nodemcu-tool mkfs --port=COM3
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool] Unable to establish connection - Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
```

I have installed Lua:
```
C:\Users\Konrad>lua -v
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio
```

What may be the reason for that?

AndiDittrich, 2017-09-25:
can you access the NodeMCU module via serial terminal ?

nodemcu-tool doesn't rely on any lua interpreter on the client side. it is a general communication error between the module and your pc. maybe a [reset-on-connect](https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/Reset_on_Connect.md) issue

KonradKlimczak, 2017-09-25:
I just got Putty and tried to connect to NodeMCU, but it does not respond::
[image]
[image]

I also tried your [reset-on-connect](https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/Reset_on_Connect.md) workaround: 
```
C:\Users\Konrad>nodemcu-tool mkfs --port=COM3 --connection-delay 500 --baud 9600
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool] Unable to establish connection - Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
```

Could it be that something's wrong with my NodeMCU?
I got second one (the same one) in case I'll destroy the first. I can try to use it.

AndiDittrich, 2017-09-25:
it seems to be an issue with your module. try to flash another firmware file

KonradKlimczak, 2017-09-25:
How can I do it? Where can I get it?

AndiDittrich, 2017-09-25:
https://nodemcu.readthedocs.io/en/master/en/flash/
https://nodemcu-build.com/

KonradKlimczak, 2017-09-25:
I tried two different firmware files, but still no good 😿 

```
C:\Users\Konrad\Downloads>esptool.py --port COM3 write_flash 0x00000 nodemcu-master-8-modules-2017-09-25-19-36-18-float.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 420816 bytes to 271018...
Wrote 420816 bytes (271018 compressed) at 0x00000000 in 24.0 seconds (effective 140.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

C:\Users\Konrad\Downloads>nodemcu-tool mkfs --port=COM3 --connection-delay 300
[NodeMCU-Tool] Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool] Unable to establish connection - Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
```

AndiDittrich, 2017-09-25:
try to establish a connection with putty first

1. try 9600/115200 baud
2. do a "hard reset" unplugging the module
3. you need to type some chars to get a valid response from the module (baud detection)

KonradKlimczak, 2017-09-25:
When I connect via Putty and click button next to "RST" ( I assume Reset ). Some characters appear on putty terminal:
[image]

I think it's proof that I'm connected.

I can even type in that terminal, but characters are pretty random and does not reflects the keys I use on my keyboard. ( I use Polish keyboard if it matters )

AndiDittrich, 2017-09-25:
seems to be a wrong baudrate.
did you tried 9600/19200/56700/115200 ?

the initial "boot up" notification uses a different one compared to the lua interpreter.

KonradKlimczak, 2017-09-25:
```
NodeMCU custom build by frightanic.com
        branch: master
        commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
        SSL: false
        modules: file,gpio,http,net,node,tmr,uart,wifi
 build  built on: 2017-09-25 19:35
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua
>
```
AFTER 7 HOURS!!! I AM SO HAPPY 😹 😹 😹 😹 😹 😹 😹 😹
Thank you so much!

Proper baudrata was 115200

closed 2017-09-26 by AndiDittrich