NodeMCU-Tool/NodeMCU-Tool #46: ESP32 support?

Spiritdude, 2018-01-26:
I tried to `upload` files with `nodemcu-tool` to a Wemos LOLIN32 Lite, and it failed:
```
[NodeMCU-Tool] Unable to establish connection - No response detected - is NodeMCU online and the Lua interpreter ready ?
```

When I use `terminal` command, and I type in a LUA code, then my typing doesn't show up, but when I hit RETURN the line I typed and the response show up.

AndiDittrich, 2018-01-27:
Hi Rene,
i didn't own an esp32 based board yet - therefore i cannot confirm that it is working.

regarding your issue: nodemcu-tool requires an enabled cli-echo - it will parse the response and validates it against the transmitted data. in case it is turned off (seems to be a firmware issue!) it cannot work.
please try a different firmware - maybe it will work 

best regards, Andi

Spiritdude, 2018-01-27:
Hi @AndiDittrich, I forgot to mention I used dev-esp32 NodeMCU firmware at https://github.com/nodemcu/nodemcu-firmware/tree/dev-esp32. It's the only NodeMCU LUA firmware I'm aware of (non LUA like Mongoose OS with JavaScript I tried successful).

I really like `nodemcu-tool` as it has been quite reliable (with ESP8266) compared to other tools which also suffer support for ESP32.

In order to upload LUA code I used http://chilipeppr.com/esp32 WebIDE (uses serial-json bridge in the background) but I like to stick with CLI.

Spiritdude, 2018-01-31:
I noticed `node.info()` doesn't exist yet for ESP32 NodeMCU (dev-esp32 documentation is misleading), which you seem to probe. 
```
[NodeMCU-Tool] Unable to establish connection - Invalid node.info() Response: stdin:1: attempt to call field 'info' (a nil value)
```
so it could actually work if you were probing something else. The info you get from `node.info()` is needed further?

Btw, I use following code to determine ESP8266 vs ESP32:
```lua
arch = string.match(node.chipid(),"^0x") and 'esp32' or 'esp8266'
```
as `node.chipid()` with ESP32 reports the hex direct with `0x` as prefix, ESP8266 reports the decimal.

AndiDittrich, 2018-02-14:
@Spiritdude 
i've merged your pull-request and tested the changes on a esp32 dev board. the results are very annoying...the esp32/nodemcu support is very buggy. at the moment a won't spend any afford on the development until the firmware is on a stable level.