NodeMCU-Tool/NodeMCU-Tool #49: nodemcu-tool devices not working (Windows)

alexandruantochi, 2018-06-01:
running either `nodemcu-tool devices`  or `nodemcu-tool devices --port=COM#` results in `[NodeMCU] No Connected Devices found | Total: 0`

I managed to navigate up to `
function list(){
    return _serialport.list();
}`  but I don't understand why the function calls itself.

AndiDittrich, 2018-06-01:
hi @alexandruantochi ,

please check the docs: https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/CommandReference.md#show-connected-nodemcu-modules

alexandruantochi, 2018-06-01:
The `--all` option works, but it lists all ports (including mouse for example). If I use `--json` I get an empty list `[]` and without any option I would get that there are no connected devices. Shouldn't the command work without the `--all` option and output only NodeMCU devices?

AndiDittrich, 2018-06-02:
it's the intended behaviour: the NodeMCU modules are produced by different vendors...therefore the USB VID/PID will also be different.

NodeMCU Tool contains a [whitelist](https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/lib/connector/list-devices.js) with some USB vendor IDs used by the "official" NodeMCU modules. you can create a pull request and add addtional modules :+1: