NodeMCU-Tool/NodeMCU-Tool #51: NodeMCU-Tool broken on macOS

Sosukodo, 2018-06-13:
macOS 10.13.4
npm 6.1.0 (v5 fails too)

> sudo npm install nodemcu-tool -g

>nodemcu-tool

`/opt/local/lib/node_modules/nodemcu-tool/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/build/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/build/Debug/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/build/Release/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/out/Debug/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/Debug/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/out/Release/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/Release/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/build/default/serialport.node
 → /opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/compiled/8.11.2/darwin/x64/serialport.node
    at bindings (/opt/local/lib/node_modules/nodemcu-tool/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/local/lib/node_modules/nodemcu-tool/node_modules/serialport/lib/serialport.js:12:25)`

AndiDittrich, 2018-06-14:
HI @Sosukodo ,

please try it without the `-g` flag...there are a several issue with the `serialport` component when trying to install in globally.

br, And

AndiDittrich, 2018-06-14:
another solution is to use `yarn` as node package manager instead of npm

`yarn global add nodemcu-tool`

the binary is locate in `~/.yarn/bin/nodemcu-tool`

Sosukodo, 2018-06-14:
> there are a several issue with the serialport component when trying to install [it] globally
Could you elaborate or provide links so that I may study this problem? Thanks

AndiDittrich, 2018-06-14:
https://github.com/node-serialport/node-serialport