NodeMCU-Tool/NodeMCU-Tool #52: Cannot install on windows 10

CyanoFresh, 2018-07-09:
Console:
```cmd
C:\Users\Alex>npm install nodemcu-tool -g
C:\Users\Alex\AppData\Roaming\npm\nodemcu-tool -> C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\bin\nodemcu-tool.js

> serialport@4.0.7 install C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ 'C:\\Program Files\\nodejs\\node.exe',
node-pre-gyp verb cli   'C:\\Users\\Alex\\AppData\\Roaming\\npm\\node_modules\\nodemcu-tool\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@10.5.0 | win32 | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release\serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-win32-x64.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-win32-x64.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.5.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-win32-x64.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.5.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Connection closed while downloading tarball file
node-pre-gyp verb command build [ 'rebuild' ]
Traceback (most recent call last):
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\gyp_main.py", line 16, in <module>
    sys.exit(gyp.script_main())
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 545, in script_main
    return main(sys.argv[1:])
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 538, in main
    return gyp_main(args)
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\__init__.py", line 523, in gyp_main
    generator.GenerateOutput(flat_list, targets, data, params)
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\generator\msvs.py", line 2025, in GenerateOutput
    version=msvs_version)
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\MSVSNew.py", line 213, in __init__
    self.Write()
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\MSVSNew.py", line 340, in Write
    f.close()
  File "C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\gyp\pylib\gyp\common.py", line 393, in close
    os.rename(self.tmp_path, filename)
WindowsError: [Error 183] ═хтючьюцэю ёючфрЄ№ Їрщы,
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\configure.js:336:16)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Alex\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=C:\\Users\\Alex\\AppData\\Roaming\\npm\\node_modules\\nodemcu-tool\\node_modules\\serialport\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\Alex\\AppData\\Roaming\\npm\\node_modules\\nodemcu-tool\\node_modules\\serialport\\build\\Release" "--python=python2.7" "--msvs_version=2015"
gyp ERR! cwd C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
gyp ERR! node -v v10.5.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release --python=python2.7 --msvs_version=2015' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:961:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
node-pre-gyp ERR! System Windows_NT 10.0.17134
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Alex\\AppData\\Roaming\\npm\\node_modules\\nodemcu-tool\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
node-pre-gyp ERR! node -v v10.5.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release --python=python2.7 --msvs_version=2015' (1)
npm ERR! code ELIFECYCLE
[... 672 lines cut ...]
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:961:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
node-pre-gyp ERR! System Windows_NT 10.0.17134
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Alex\\AppData\\Roaming\\npm\\node_modules\\nodemcu-tool\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
node-pre-gyp ERR! node -v v10.5.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\Alex\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\build\Release' (1)
npm WARN rollback Rolling back http-signature@1.1.1 failed (this is probably harmless): EPERM: operation not permitted, lstat 'C:\Users\Alex\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport\node_modules\node-pre-gyp\node_modules\request\node_modules'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Alex\AppData\Roaming\npm-cache\_logs\2018-07-09T13_45_16_708Z-debug.log
```
Here is log
[2018-07-09T13_45_16_708Z-debug.log](https://github.com/AndiDittrich/NodeMCU-Tool/files/2176408/2018-07-09T13_45_16_708Z-debug.log)

AndiDittrich, 2018-07-09:
Hi @CyanoFresh ,

the issue is related to the node-serialport package.

did you try to install NodeMCU-Tool without the global flag ?

CyanoFresh, 2018-07-09:
thanks for reply. same errors

AndiDittrich, 2018-07-09:
maybe it's the serialport v4 package. the NodeMCU-Tool master branch uses 6.2 currently - i will push a tag soon

AndiDittrich, 2018-07-09:
i've just released NodeMCU-Tool **v3** including serialport **v6.2** - this should fix the issue