NodeMCU-Tool/NodeMCU-Tool #53: Mac OSX serial port problem

arrowcircle, 2018-07-22:
Hi!
On Mac OSX nodemcu devices appear not as `/dev/ttyUSB0`, but as `/dev/tty.SLAB_USBtoUART`.

Here is two outputs:

```
./nodemcu-tool devices
[NodeMCU-Tool]~ Unable to establish connection
[NodeMCU-Tool]~ Cannot open port "/dev/ttyUSB0"
```

```
nodemcu-tool devices --port /dev/tty.SLAB_USBtoUART
[NodeMCU-Tool]~ Connected
[device]      ~ Arch: esp8266 | Version: 2.2.0 | ChipID: 0x3a7c4f | FlashID: 0x1640ef
[device]      ~ Connected Devices | Total: 1
[device]      ~ - /dev/tty.SLAB_USBtoUART (Silicon Labs, undefined)
[NodeMCU-Tool]~ disconnecting
```

How it could be fixed?

AndiDittrich, 2018-07-22:
Hi @arrowcircle ,

hmmm it shouldn't open a port to list devices...i will check this.

best regards, Andi

AndiDittrich, 2018-07-22:
HI @arrowcircle ,

thanks for your report! it is fixed in **3.0.1** (current master branch)

background: during the refactoring of the codebase i've used some special commands for testing...therefore the `devices` action tried to establish a serial connection which is not required by the command.

sorry for this!

best regards, Andi

closed 2018-07-22 by AndiDittrich

AndiDittrich, 2018-07-22:
v3.0.1 has been release (GitHub + NPM)

commit b6c488ab (2018-07-22): fixed #53 - devices command did not require a valid connection