NodeMCU-Tool/NodeMCU-Tool #54: Unable to establish connection - No response detected

lucaszanella, 2018-09-25:
I have a brand new nodemcu. I followed https://nodemcu.readthedocs.io/en/master/en/getting-started/ and built using this dockerized build: https://github.com/marcelstoer/docker-nodemcu-build

I can see the mcu using the `devices` option, but this is what I get when I try to clean the filesystem:

> [NodeMCU-Tool] Unable to establish connection - No response detected - is NodeMCU online and the Lua interpreter ready ?

The error happens right after the echo

I've found https://github.com/AndiDittrich/NodeMCU-Tool/issues/46 but it's for another ESP, not mine. 

By the way, could it be the baud rate? Which is the default for this tool? I think for the firmware it is 115200.

AndiDittrich, 2018-09-25:
hi @lucaszanella ,

1. does the module works within a terminal ?
2. did you select the correct device ?
3. the issue can be caused by an invalid baudrate but the current NodeMCU releases uses an autobaud detection. you can also change the baudrate of NodeMCU-Tool.
4. check if its a [reset-on-connect](https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/Reset_on_Connect.md) issue

lucaszanella, 2018-09-26:
1 )When I do screen /dev/ttyUSB0 155200 I can send things to the nodemcu. If I do it without 115200 whenever I type soething, random characters appear on screen. With 115200 normal characters appear but I don't know what to type to talk with it. 

2) I selected the correct device because it's the only one that appears when I run the command `devices`

3 )

```
lz@vm:~/node-mcu-test$ ./node_modules/nodemcu-tool/bin/nodemcu-tool.js mkfs --port=/dev/ttyUSB0  --baud 115200
[NodeMCU-Tool]~ Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Cannot open port "/dev/ttyUSB0" 

```

4) 

```
lz@vm:~/node-mcu-test$ ./node_modules/nodemcu-tool/bin/nodemcu-tool.js mkfs --port=/dev/ttyUSB0 --connection-delay 400 --baud 115200
[NodeMCU-Tool]~ Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Cannot open port "/dev/ttyUSB0" 

```

lucaszanella, 2018-09-26:
Actually I just noticed the error I posted here wasn't the same as I posted before. Now it was an error due to another thing. Anyways, I did now with connection delay and it worked

closed 2018-09-26 by AndiDittrich