NodeMCU-Tool/NodeMCU-Tool #56: upload/download error when nodemcu is printing information

LanFly, 2018-11-07:
hi, this tool is very useful to me.
But I am having a problem when using this tool.

for example:
I print a line of information every second with the print statement in nodemcu.
Then I tried to upload a file but it failed. The failed message is what I printed with print.

I am also worried that if I just print a line of information during the process of downloading the file, the content of the downloaded file is wrong.

I tested this case in ESPlorer, it works fine. I am looking at how ESPlorer does it.

AndiDittrich, 2018-11-07:
Hi @LanFly ,

workaround: don't use print statements on the serial interface used for uploading.. :stuck_out_tongue: 

i'm not familiar with ESPlorer but i believe that it disables standard `print` commands while it interacts with the module.

there is no generic solution for this problem...

best regards, Andi

LanFly, 2018-11-07:
thank you for your reply！
Can you tell me how to turn off and turn on the `print` commands?
Because I need to upload large files, this may take some time, I want to make sure that the upload function will not be affected by print.

AndiDittrich, 2018-11-07:
you should replace `print` with a custom "debug" functions which can be enabled via a global flag.
on before uploading/downloading code you can run a command via NodeMCU tools which "disables" the global flag

closed 2018-11-09 by LanFly