NodeMCU-Tool/NodeMCU-Tool #57: Unable to establish connection - timeout

jonascoenen005, 2018-12-18:
Hello,

I'm trying to upload init.lua file with the nodeMCU-tool with the following command:
`nodemcu-tool upload --port=/dev/tty.wchusbserial1430 init.lua`

I'm getting the following error: 

> [NodeMCU-Tool]~ Unable to establish connection 
> [NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 

I already tried adding --connection-delay 50, didn't work
I tried reflashing my ESP8266, didn't work.

How can i find the issue?

jonascoenen005, 2018-12-18:
I have tried the nodemcu terminal mode by doing nodemcu-tool terminal --port=/dev/tty.wchusbserial1430 and it hangs.

AndiDittrich, 2018-12-18:
please try a standard serial terminal program like putty, hterm, minicom

LanFly, 2018-12-25:
```json
{
    "baudrate": "115200",
    "port": "/dev/tty.SLAB_USBtoUART",
    "minify": false,
    "compile": false,
    "keeppath": true
}
```
Have you tried setting the baud rate to 9600 or 115200? Save the above to the file `.nodemcu` and try again `nodemcu-tool upload init.lua`.

AsyDynamics, 2019-01-09:
Getting same error, using firmware built online and luanode based on esp32. Tried solutions mentioned above but still not work

jonascoenen005, 2019-01-09:
Sorry for the late response, but i didn't try it anymore because i'm using ArduinoIDE to write/upload my scripts. However i also had an issue with the arduinoIDE, but it went away when i made a new NodeMCU custom build from the DEV branch (https://nodemcu-build.com/) perhaps you can try that

AndiDittrich, 2019-01-09:
@AsyDynamics 
your issue seems to be esp32 related - not esp8266 (classic nodemcu lua) - it is currently in a very experimental state. please open a new issue to discuss your problem

micheldrescher, 2019-01-10:
Hi, I saw a very similar issue on StackOverflow (https://stackoverflow.com/questions/53624685/cant-upload-to-the-nodemcu-lua). I had a very similar (if not identical) issue, and responded there. But I am happy to summarise here:

Your port seem to indicate you are running on a Mac, as I do, and the port where my ESP8266 (Wemos D1) was connected to looks very, very similar.

I had similar issues, which seem related to the unreliable way of how Node.js and particularly nodemcu-tool are installed on Mac OS X systems. Long story short, the way I had to invoke nodemcu-tool was to "sudo" it, f.i. "sudo nodemcu-tool fsinfo".

That is not desirable. The only way I was able to get past the problems was to:
- Completely remove Node.js and start from scratch (perhaps not necessary)
- Completely remove nodemcu-tool
- Install Node.js again as per instructions
- Install nodemcu-tool but with one *important* addition: Include param "--unsafe-perm" as a global installation, i.e. "sudo npm install --unsafe-perm nodemcu-tool -g"

Then I was able to use --connection_delay to connect to my ESP8266.

AsyDynamics, 2019-01-10:
thanks for your info, and I was running on linux Mint. I just switched to
arduino ide and it worked fine. Later I would try your sudo instructions.

dacodekid, 2019-01-16:
@AndiDittrich I can confirm that this is happing on `esp8266` - tested on couple modules with no luck.

AndiDittrich, 2019-01-17:
@dacodekid 

the "issue" is not as simple and may not related to NodeMCU-Tool itself....it's a very generic error message which can have **multiple causes** depending on the hardware/serial drivers/...

therefore, i will **close this issue right now** - please open a [custom report](https://github.com/AndiDittrich/NodeMCU-Tool/issues/new) regarding your issue

best regards, Andi

closed 2019-01-17 by AndiDittrich

ideal, 2021-06-20:
[image]

I encountered the same problem on Mac OS with NodeMCU V1.2 ESP32.
Has someone solved it? Thanks.