NodeMCU-Tool/NodeMCU-Tool #59: Unable to unstall on windows 10

IDispose, 2019-01-14:
On Windows 10
Node version 11.0.0
npm version 6.4.1

Trying to install via npm install nodemcu-tool -g throws the following error

```
C:\Users\xxx\AppData\Roaming\npm\nodemcu-tool -> C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\bin\nodemcu-tool.js

> serialport@6.2.2 install C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=11.0.0 runtime=node arch=x64 platform=win32)

C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:484:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:509:16)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqCallback.oncomplete (fs.js:161:21)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
gyp ERR! node -v v11.0.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\xxx\AppData\Roaming\npm-cache\_logs\2019-01-14T00_54_15_295Z-debug.log

```
If this is a nodejs program why does it need Python?

So just to humor the install, tried via Anaconda prompt and got the following 

```
C:\Users\xxx\AppData\Roaming\npm\nodemcu-tool -> C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\bin\nodemcu-tool.js

> serialport@6.2.2 install C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=11.0.0 runtime=node arch=x64 platform=win32)

(base) C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Command failed: C:\ProgramData\Anaconda3\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                                ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:289:12)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\xxx\AppData\Roaming\npm\node_modules\nodemcu-tool\node_modules\serialport
gyp ERR! node -v v11.0.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\xxx\AppData\Roaming\npm-cache\_logs\2019-01-14T01_03_07_635Z-debug.log

```
Tried logged in as an administrator. No luck. Tied on a Raspberry Pi. Same errors. 

Help please

AndiDittrich, 2019-01-14:
@IDispose 

the issue is not related to nodemcu-tool itself.... its a problem with one of the dependencies [node-serialport](https://github.com/node-serialport/node-serialport) which is required to access the serial hardware by using native/os dependent code.

normally the package will use prebuild binaries to provide the additional functionality (hardware access) to node.js - in case the binaries are not available it tries to **build** them on the current system by invoking `node-gyp` which requires python....

i will take a look on it why the prebuild binaries are not used...

did you try do install it without the `-g` flag ?

best regards, Andi

IDispose, 2019-01-14:
@AndiDittrich 
Thanks for the clarification on use of Python. 

Yes, I did try -g flag as well with same results. I'll roll back node's to V10 LTS and try.

IDispose, 2019-01-15:
@AndiDittrich 

Uninstalled Node 11.0.0 and installed 10.15.0 LTS.
Was able to install nodemcu-tool via npm no errors. and it works. 

THANK YOU!

closed 2019-01-15 by IDispose