NodeMCU-Tool/NodeMCU-Tool #63: Can't connect to NodeMCU after flashing firmware.

nolash, 2019-05-13:
## Checklist ##

1. Can you establish a serial connection **WITHOUT** Nodemcu-Tool - e.g. with a serial terminal like Putty ?  
  **YES/NO**  

YES to /dev/ttyUSB0 but it outputs a bunch of binary garbage:

```
# cat /dev/ttyUSB0 | hexdump
0000000 434f 9245 4305 4f04 7141 5206 fc52 8230
0000010 3f7e 3496 5293 12b6 ffa3 4f04 2ab4 1ab6
0000020 0443 434f 9245 4305 4704 7151 5206 fc52
0000030 8230 3f7e 2486 4293 12b6 ffa3 5f14 3aa4
0000040 1ab6 0443 434f 9245 4305 4f04 7141 5206
0000050 fc52 8230 3f7e 3496 5293 12b6 ffa3 4f04
0000060 2ab4 1ab6 0443 434f 9245 4305 4f04 7141
0000070 4206 fc52 8230 3f7e 3496 5293 02a6 ffa3
0000080 4f04 2ab4 1ab6 044b 434f 9245 4305 4f04
0000090 7941 5206 fc52 8230 3f7e 3496 4293 12b6
00000a0 ffa3 5f14 3ab4 1ab6 0443 434f 8245 4305
00000b0 4f04 7141 5206 fc52 8230 3f7e 3496 5293
00000c0 12b6 ffa3 4f04 3ab4 1ab6 0443 434f 924d
00000d0 4305 4f04 7141 5206 fc52 8230 3f7e 3496
00000e0 5293 12b6 ffa3 5f14 3ab4 1ab6 0443 434f
00000f0 924d 4305 4f04 7141 5206 fc52 8220 3f6e
0000100 3496 4293 12b6 ffa3 4f04 3ab4 1ab6 0443
0000110 434f 9245 4305 4f04 7141 5206 fc52 8230
0000120 3f7e 3496 5293 12b6 ffa3 4f04 3ab4 1ab6
0000130 0443 434f 9245 4305 4f04 7141 5206 fc52
0000140 8220 1f4e 2486 5293 0286 ffa3 4f04 3ab4
0000150 1ab6 1443 435f 9245 4305 4f04 7141 4206
0000160 fc42 8230 1f76 3496 4293 12b6 ffa3 4f04
0000170 3ab4 1ab6 0443 434f 9245 4305 4f04 7141
```  

2. Does the `nodemcu-tool fsinfo` command work ?  

NO

-------------

## Environment ##

* **Operating System**  
 
Linux sostenuto 5.0.8-arch1-1-nousbserial #1 SMP PREEMPT Fri May 10 00:18:58 CEST 2019 x86_64 GNU/Linux

* **ESP Device/Revision**  

```
$ lsusb -v -d 10c4:ea60

Bus 002 Device 071: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP2102/CP2109 UART Bridge Controller [CP210x family]
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

* **NodeMCU-Tool Version**  
  Please use the least recent version to validate the issue  
  `3.0.2`
* **Node.js Version**  

`v11.15.0`

* **NodeMCU LUA Firmware Version**  
  `1.5.4`

-------------

## Debug Output ##

```
$ sudo node bin/nodemcu-tool.js -p /dev/ttyUSB0 --debug --connection-delay 1000 mkfs
[config]      ~ debug mode enabled 
[config]      ~ Project based configuration loaded 
[config]      ~ baudrate             = 115200 (type:string) 
[config]      ~ port                 = /dev/ttyUSB0 (type:string) 
[config]      ~ connectionDelay      = 1000 (type:string) 
[config]      ~ minify               = false (type:boolean) 
[config]      ~ compile              = false (type:boolean) 
[config]      ~ keeppath             = false (type:boolean) 
[config]      ~ remotename           = null (type:object) 
[config]      ~ run                  = false (type:boolean) 
[config]      ~ all                  = false (type:boolean) 
[config]      ~ json                 = false (type:boolean) 
[config]      ~ raw                  = false (type:boolean) 
[config]      ~ softreset            = false (type:boolean) 
[NodeMCU-Tool]~ Do you really want to format the filesystem and delete all file ? (no) yes
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 
[NodeMCU-Tool]~ Error: Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
    at Timeout._onTimeout (/home/lash/bin/nodemcu/node_modules/nodemcu-tool/lib/connector/check-connection.js:13:20)
    at listOnTimeout (internal/timers.js:535:17)
    at processTimers (internal/timers.js:479:7) 

```
$ lsmod | grep usb
btusb                  57344  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                28672  1 btusb
bluetooth             651264  4 btrtl,btintel,btbcm,btusb
usbserial              57344  1 cp210x
usb_storage            73728  1 uas
usbhid                 61440  1 wacom
hid                   147456  3 wacom,usbhid,hid_generic
scsi_mod              253952  5 sd_mod,usb_storage,uas,libata,sg
```

AndiDittrich, 2019-05-13:
Hi @nolash 

your issue is not related to NodeMCU-Tool...it's related to your firmware because you cannot establish a working connection (this assumes the serial transport as well as the terminal is functional).

maybe it's a baudrate related issue

nolash, 2019-05-13:
Oh well yeah it seems this unit has its own software actually. Sorry ;/

closed 2019-05-13 by nolash