NodeMCU-Tool/NodeMCU-Tool #64: npm install fails on Ubuntu 18

jkobie, 2019-07-03:
NVM 
nvm --version
0.34.0

node --version

v12.1.0
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic

to reproduce error:
git clone https://github.com/AndiDittrich/NodeMCU-Tool.git
cd NodeMCU-Tool
npm install

> serialport@6.2.2 install /home/jkobie/work/NodeMCU-Tool-3.0.2/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.1.0 runtime=node arch=x64 platform=linux)
make: Entering directory '/home/jkobie/work/NodeMCU-Tool-3.0.2/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:41:48: error: no matching function for call to ‘v8::Value::ToString()’
   v8::String::Utf8Value path(info[0]->ToString());
                                                ^
In file included from /home/jkobie/.node-gyp/12.1.0/include/node/node.h:63:0,
                 from ../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/jkobie/.node-gyp/12.1.0/include/node/v8.h:2528:44: note: candidate: v8::MaybeLocal<v8::String> v8::Value::ToString(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
                                            ^~~~~~~~
/home/jkobie/.node-gyp/12.1.0/include/node/v8.h:2528:44: note:   candidate expects 1 argument, 0 provided
In file included from /home/jkobie/.node-gyp/12.1.0/include/node/v8-internal.h:14:0,
                 from /home/jkobie/.node-gyp/12.1.0/include/node/v8.h:25,
                 from /home/jkobie/.node-gyp/12.1.0/include/node/node.h:63,
                 from ../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/jkobie/.node-gyp/12.1.0/include/node/v8.h:2544:35: note: candidate: v8::Local<v8::String> v8::Value::ToString(v8::Isolate*) const
                     Local<String> ToString(Isolate* isolate) const);
                                   ^
/home/jkobie/.node-gyp/12.1.0/include/node/v8config.h:322:3: note: in definition of macro ‘V8_DEPRECATE_SOON’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
/home/jkobie/.node-gyp/12.1.0/include/node/v8.h:2544:35: note:   candidate expects 1 argument, 0 provided
                     Local<String> ToString(Isolate* isolate) const);
                                   ^
/home/jkobie/.node-gyp/12.1.0/include/node/v8config.h:322:3: note: in definition of macro ‘V8_DEPRECATE_SOON’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/serialport.cpp:48:53: error: no matching function for call to ‘v8::Value::ToObject()’
   v8::Local<v8::Object> options = info[1]->ToObject();
                                                     ^
In file included from /home/jkobie/.node-gyp/12.1.0/include/node/node.h:63:0,
                 from ../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/jkobie/.node-gyp/12.1.0/include/node/v8.h:2532:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Value::ToObject(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                            ^~~~~~~~
/home/jkobie/.node-gyp/12.1.0/include/node/v8.h:2532:44: note:   candidate expects 1 argument, 0 provided
In file included from /home/jkobie/.node-gyp/12.1.0/include/node/v8-internal.h:14:0,
                 from /home/jkobie/.node-gyp/12.1.0/include/node/v8.h:25,
                 from /home/jkobie/.node-gyp/12.1.0/include/node/node.h:63,
                 from ../../nan/nan.h:54,
[... 241 lines cut ...]
 ^~~~~~~~~~~
serialport.target.mk:110: recipe for target 'Release/obj.target/serialport/src/serialport.o' failed
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory '/home/jkobie/work/NodeMCU-Tool-3.0.2/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/jkobie/.nvm/versions/node/v12.1.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:196:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:256:12)
gyp ERR! System Linux 4.18.0-25-generic
gyp ERR! command "/home/jkobie/.nvm/versions/node/v12.1.0/bin/node" "/home/jkobie/.nvm/versions/node/v12.1.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/jkobie/work/NodeMCU-Tool-3.0.2/node_modules/serialport
gyp ERR! node -v v12.1.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/jkobie/.npm/_logs/2019-07-03T14_47_45_961Z-debug.log

AndiDittrich, 2019-07-03:
Hi @jkobie ,

as you can see in the error logs...the issue is caused by the nodejs serialport package which includes native code. the v6 used by NodeMCU tool is not compatible with Node.js >= 11.

you can downgrade to Node.js 10 LTS - this should work fine.

i'll check if it's possible to use serialport v7 with NodeMCU tool.

best regards, Andi

jkobie, 2019-07-03:
Thank you.

nvm install v10.16.0
nvm use v10.16.0

Resolved issue.

jkobie, 2019-07-03:
nvm install v10.16.0
nvm use v10.16.0

Resolved issue

closed 2019-07-03 by jkobie