NodeMCU-Tool/NodeMCU-Tool #65: Support for latest NodeMCU 3.o

marcelstoer, 2019-09-09:
As per https://stackoverflow.com/a/57845719/131929 the latest NodeMCU 2.2.x -> 3.0 upgrade broke the `fetchDeviceInfo()` function. `node.info()` now works differently as per https://nodemcu.readthedocs.io/en/latest/modules/node/#nodeinfo.

AndiDittrich, 2019-09-09:
Hi @marcelstoer ,

the issue seems only be related to the nodemcu-firmware >= 2.x.

i will check if it's possible to add some other kind of "detection"

marcelstoer, 2019-09-09:
> the issue seems only be related to the nodemcu-firmware >= 2.x.

Yes. That's why I think the title of this issue is now wrong.

AndiDittrich, 2019-09-09:
not really... nodemcu-firmware is on [2.2.1](https://github.com/nodemcu/nodemcu-firmware/releases) which causes the issue..

marcelstoer, 2019-09-09:
We are both right - kind of. As I said we upgraded to SDK 3.0 i.e. we snapped our `dev` branch back to `master` just 2d ago.  However, the release tag and the notes are pending. Will have to do that later today.

AndiDittrich, 2019-09-09:
ok sry my fault.

AndiDittrich, 2019-09-09:
fixed within the master branch - projected for 3.1.0

alset333, 2019-10-13:
Anticipated timeframe? This still seems to be unchanged in npm and trying to build is causing headaches personally.

AndiDittrich, 2019-10-13:
i didn't noticed that there is a stable release of 3.x out - i'll release the changes asap

AndiDittrich, 2019-10-13:
tagged 3.1.0 and published to npm

closed 2019-10-13 by AndiDittrich