NodeMCU-Tool/NodeMCU-Tool #66: Glob arguments are not expanded on Windows

ivkos, 2019-11-17:
* Globs are strings like **`src/*.lua`** that match multiple files
* Linux bash-like shells expand them into a list of files when invoking a command, i.e. invoking this:
  - **`nodemcu-tool upload --minify --compile src/*.lua`** actually executes this:
  - **`nodemcu-tool upload --minify --compile src/init.lua src/my_script.lua`**
* Windows shells (cmd.exe, PowerShell) do not do that

So, in short, the issue is that executing the same command in bash works fine, and fails in Windows shells due to the lack of glob expansion. This makes it hard to set up some useful scripts in my `package.json` that work on all platforms.

A possible solution would be parsing the CLI arguments and checking with **[is-glob](https://www.npmjs.com/package/is-glob)** whether the arg looks like a glob, then resolving it with **[glob](https://www.npmjs.com/package/glob)** into a list of files.

AndiDittrich, 2019-11-17:
Hi @ivkos,

i can add this as features request but i didn't have the time to implement it soon - maybe in the future..

ivkos, 2019-11-19:
Thanks Andi. I'll see if I can help with this if I find some spare time.

AndiDittrich, 2019-11-19:
i've just added a [wrapper](https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/lib/cli/glob-expression.js) to invoke the glob expansion for the upload command

the expansion code is still empty (passthrough)

AndiDittrich, 2019-12-22:
v3.2.0 is out including generic glob resolver

closed 2019-12-22 by AndiDittrich