NodeMCU-Tool/NodeMCU-Tool #69: Cannot rename files on upload

marcoskirsch, 2019-12-23:
## Checklist ##

1. Can you establish a serial connection **WITHOUT** Nodemcu-Tool - e.g. with a serial terminal like Putty ?  
  
N/A

2. Does the `nodemcu-tool fsinfo` command work ?  

N/A

-------------

## Environment ##

N/A

-------------

## Debug Output ##

N/A

-------------

## Issue Description ##

> [nodemcu-httpserver](https://github.com/marcoskirsch/nodemcu-httpserver/) relies on nodemcu-uploader file renaming feature in order to simulate http folder in SPIFFS. nodemcu-tool does not have this feature.

### Expected Behavior ###

It would be great to move nodemcu-httpserver to use this tool instead of the dead nodemcu-uploader. This feature is missing for a clean replacement.

See https://github.com/marcoskirsch/nodemcu-httpserver/issues/127

HHHartmann, 2019-12-23:
Uploading files has an option --keeppath which should do the trick.

AndiDittrich, 2019-12-23:
i've posted an example into the nodemcu-httpserver issue

see https://github.com/marcoskirsch/nodemcu-httpserver/issues/127#issuecomment-568429706

closed 2019-12-23 by AndiDittrich

AndiDittrich, 2019-12-23:
btw. by using the `--compile` option you can simplify your deployment

```
 $ ./nodemcu-tool upload --keeppath --compile srv/*
[NodeMCU-Tool]~ Connected 
[device]      ~ Arch: esp8266 | Version: 3.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0 
[NodeMCU-Tool]~ Uploading "srv/dummy_strings.lua" >> "srv/dummy_strings.lua"... 
[connector]   ~ Transfer-Mode: base64 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-b64decode.lua" >> "srv/httpserver-b64decode.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-basicauth.lua" >> "srv/httpserver-basicauth.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-buffer.lua" >> "srv/httpserver-buffer.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-connection.lua" >> "srv/httpserver-connection.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-error.lua" >> "srv/httpserver-error.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-header.lua" >> "srv/httpserver-header.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-init.lua" >> "srv/httpserver-init.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver.lua" >> "srv/httpserver.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-request.lua" >> "srv/httpserver-request.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-static.lua" >> "srv/httpserver-static.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/httpserver-wifi.lua" >> "srv/httpserver-wifi.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Uploading "srv/_init.lua" >> "srv/_init.lua"... 
[device]      ~  |- compiling lua file.. 
[device]      ~  |- success 
[device]      ~  |- original Lua file removed 
[NodeMCU-Tool]~ Bulk File Transfer complete! 
[NodeMCU-Tool]~ disconnecting 

 $ ./nodemcu-tool fsinfo
[NodeMCU-Tool]~ Connected 
[device]      ~ Arch: esp8266 | Version: 3.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0 
[device]      ~ Free Disk Space: 3420 KB | Total: 3442 KB | 13 Files 
[device]      ~ Files stored into Flash (SPIFFS) 
[device]      ~  - srv/httpserver.lc (4256 Bytes) 
[device]      ~  - srv/httpserver-b64decode.lc (1868 Bytes) 
[device]      ~  - srv/httpserver-basicauth.lc (940 Bytes) 
[device]      ~  - srv/httpserver-connection.lc (916 Bytes) 
[device]      ~  - srv/dummy_strings.lc (1016 Bytes) 
[device]      ~  - srv/_init.lc (1240 Bytes) 
[device]      ~  - srv/httpserver-static.lc (372 Bytes) 
[device]      ~  - srv/httpserver-error.lc (368 Bytes) 
[device]      ~  - srv/httpserver-buffer.lc (776 Bytes) 
[device]      ~  - srv/httpserver-header.lc (1140 Bytes) 
[device]      ~  - srv/httpserver-init.lc (1540 Bytes) 
[device]      ~  - srv/httpserver-wifi.lc (700 Bytes) 
[device]      ~  - srv/httpserver-request.lc (2576 Bytes) 
[NodeMCU-Tool]~ disconnecting 

```