NodeMCU-Tool/NodeMCU-Tool #71: [Feature Request] Compile files before upload

jayu, 2020-03-14:
## Background
Hey, this tool looks amazing! I'm doing a small research about upgrading my current dev tooling around NodeMCU development.

I used to use ESPlorer, but I'm looking for a more convenient and automated way of uploading and executing scripts on my ESPs. Recently I faced an issue with hight ram usage, due to runtime compilation of '*.lua' files. Due to [docs](https://nodemcu.readthedocs.io/en/master/compiling/#compiling-lua-on-your-pc-for-uploading), it is possible to compile source files on PC, before uploading them to ESP. It should save a lot of RAM which is being used to compile files to byte code.

## Feature Request
Compiling the code is not yet supported by this tool. Of course, I can first compile the files using `luac.cross` and then upload them with `NodeMCU-Tool`, but It would be more convenient to have this feature built-in.

We could introduce an additional params `-c` and `--compiler` to `upload` command that would be used to point `luac.cross` binary. Then the tool could use this binary to compile provided `*.lua` files before upload. But it would require checking the files extensions first and might not be intuitive for users.

Another option is to add a new command, e.g `compile-upload` with `-c` and `--compiler`  params that would be only for `*.lua` sources. But this has a drawback, that some users might want to upload not only the `*.lua` files.

Some users might want to pass additional params to the compiler. E.g. it is possible to build one `*.img` from many `*.lua` source files. But it would not fit into any of the above options since we are changing the files that would be uploaded (before we had to just rename them from `*.lua` to `*.lc`).

So there is one more possibility. Provide a wrapper over `luac.cross` that would take a list of files and compile them, and then print paths to files to stdout. Then we could pipe `compile` and `upload`
using `xargs` e.g `node-tool compile -c [compiler path] file.lua file2.lua | xargs node-tool upload`.
It could be used in both scenarios, meaning whether the user wants to bundle files into an image or not. 

## Current solution

I found out that there is an option to compile a file on ESP, but if the file is too big it will run out of memory. However, the message is still `success`. I will open another issue for that. 
Compiling on PC will allow handling bigger files and have more source code on ESP.

AndiDittrich, 2020-03-14:
Hi @jayu ,

the main issue with such feature is that most nodemcu users are not able to install/setup a full toolchain (which is also platform depended) - therefore i've given up this idea a long time ago.

currently i didn't have any time to implement this (or a similar) feature...sry

jayu, 2020-03-14:
I'm totally fine with that. This feature could be usefull for more advance ppl. Regular users might not have to use it. 
And I forgot to mention that I would be happy to contribute! Let me know if you see a place for that kind of feature in this library. If we establish an API for this I can give i a try :D