NodeMCU-Tool/NodeMCU-Tool #74: Annoying warning for every command with nodemcu-tool

buzzy, 2020-05-01:
(node:7165) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency

AndiDittrich, 2020-05-01:
Hi @buzzy 

with node 12LTS i'm unable to reproduce this issue.

can you please provide more informations about your platform/versions etc ?

AndiDittrich, 2020-05-01:
it seems to be a general issue with nodejs 14 - please use the lts version

buzzy, 2020-05-01:
Well, more than me will have the issue. I installed node with "brew install node", like most people will on Mac. Anyways, thanks for the quick response!

closed 2020-05-01 by buzzy

jamsinclair, 2022-05-16:
The error was:
```
(node:1793) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
    at emitCircularRequireWarning (node:internal/modules/cjs/loader:707:11)
    at Object.get (node:internal/modules/cjs/loader:721:5)
    at Object.exports.setLevels (/Users/person/workspace/NodeMCU-Tool/node_modules/winston/lib/winston/common.js:32:14)
    at Object.<anonymous> (/Users/person/workspace/NodeMCU-Tool/node_modules/winston/lib/winston.js:83:8)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
```

This was caused by an old version of `winston` in use in the `prompt` dependency.

Upgrading the project's `prompt` dependency to latest version seems to resolve this.