NodeMCU-Tool/NodeMCU-Tool #77: [NodeMCU-Tool]~ Unable to establish connection

CSV-12, 2020-05-31:
------------
Not sure how to deal with NodeMCU LUA 
## Checklist ##

1. Can you establish a serial connection **WITHOUT** Nodemcu-Tool - e.g. with a serial terminal like Putty ?  
  **YES**  
  If not, the issue not related to NodeMCU-Tool - it might be a problem with your serial driver, NodeMCU hardware and/or firmware  

2. Does the `nodemcu-tool fsinfo` command work ?  
  **NO**
/node_modules/nodemcu-tool/bin $ nodemcu-tool fsinfo
(node:10937) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 

-------------

## Environment ##

* **Operating System**  
  / Ubuntu 18.04.1 LTS / Mint Sylvia 
* **ESP Device/Revision**  
  ESP8266
  Original NodeMCU `v1.1` (not sure, got it yesterday)

* **NodeMCU-Tool Version**  
  Please use the least recent version to validate the issue  
 ~/node_modules/nodemcu-tool/bin $ ./nodemcu-tool.js --version
3.2.1
npm@6.14.5 /usr/lib/node_modules/npm

* **Node.js Version**  
  v14.3.0

* **NodeMCU LUA Firmware Version**  

(not sure)

-------------

## Debug Output ##

> **NOTE** In case you have issues with some commands of NodeMCU-Tool, please provide the output of the command including the debug flags
>
```
something like, 

(mine is):
~/node_modules/nodemcu-tool/bin $ nodemcu-tool --debug --io-debug fsinfo
[config]      ~ debug mode enabled 
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger 
(node:10657) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
[config]      ~ baudrate             = 115200 (type:string) 
[config]      ~ port                 = /dev/ttyUSB0 (type:string) 
[config]      ~ connectionDelay      = 0 (type:number) 
[config]      ~ minify               = false (type:boolean) 
[config]      ~ compile              = false (type:boolean) 
[config]      ~ keeppath             = false (type:boolean) 
[config]      ~ remotename           = null (type:object) 
[config]      ~ run                  = false (type:boolean) 
[config]      ~ all                  = false (type:boolean) 
[config]      ~ json                 = false (type:boolean) 
[config]      ~ raw                  = false (type:boolean) 
[config]      ~ softreset            = false (type:boolean) 
[config]      ~ noninteractive       = false (type:boolean) 
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: p
[serialport]  ~ RX: rint("ec
[serialport]  ~ RX: ho1337")
E
[serialport]  ~ RX: RROR
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 
[NodeMCU-Tool]~ Error: Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
    at Timeout._onTimeout (/usr/lib/node_modules/nodemcu-tool/lib/connector/check-connection.js:13:20)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7) 

-------------

## Issue Description ##

### Expected Behavior ###

...

### Current Behavior ###

...

### Steps to Reproduce ###

1. X
2. Y
3. Z

### Detailed Description ###

...

### Possible Solution ###

...

AndiDittrich, 2020-05-31:
the debug output looks weird: 

```
serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: p
[serialport]  ~ RX: rint("ec
[serialport]  ~ RX: ho1337")
E
[serialport]  ~ RX: RROR
```

what happens if you execute the command `print("echo1337")` within a terminal ?
currently i assume that your nodemcu firmware is broken...just flash a recent version

CSV-12, 2020-06-01:
"...product is sold as a NodeMCU Lua board, and while the board itself is based on the NodeMCU board, it does not include the NodeMCU Lua firmware. In fact, the firmware includes no Lua interpreter at all, just the stock AT firmware from the ESP8266 module. ..."

I have to figure how to to flash it for Lua.

AndiDittrich, 2020-06-02:
use `esptool.py` it's quite simple. i'll close this thread for now

closed 2020-06-02 by AndiDittrich