NodeMCU-Tool/NodeMCU-Tool #79: [NodeMCU-Tool]~ Unable to establish connection  [NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?

mh4nx7net, 2020-10-01:
-------------

## Checklist ##

1. i can establish my espressif serial connection, either with `screen` and `nodemcu-tool terminal`
2. `nodemcu-tool fsinfo`
```
(researchDev) ☁  research  nodemcu-tool fsinfo
[config]      ~ Project based configuration loaded
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 
```
3. `esptool.py flash_id`
```
(researchDev) ☁  research  esptool.py flash_id
esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:d6:7e:92
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

-------------

## Environment ##

* **Operating System**  
Ubuntu 18.04.1 LTS
* **ESP Device/Revision**  
  ESP8266-12E
  Original NodeMCU `v1.1`
* **NodeMCU-Tool Version**  
```
(researchDev) ☁  research  nodemcu-tool --version
3.2.1
```
* **Node.js Version**  
(researchDev) ☁  research  node --version        
v12.3.1
* **NodeMCU LUA Firmware Version**  
idk

-------------

## Debug Output ##
```
(researchDev) ☁  IOT  nodemcu-tool --debug --io-debug info
[NodeMCU-Tool]~ Unknown command "info" 
```
```
(researchDev) ☁  IOT  nodemcu-tool --debug --io-debug fsinfo
[config]      ~ debug mode enabled
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger
[config]      ~ baudrate             = 115200 (type:string)
[config]      ~ port                 = /dev/ttyUSB0 (type:string)
[config]      ~ connectionDelay      = 0 (type:number)
[config]      ~ minify               = false (type:boolean)
[config]      ~ compile              = false (type:boolean)
[config]      ~ keeppath             = false (type:boolean)
[config]      ~ remotename           = null (type:object)
[config]      ~ run                  = false (type:boolean)
[config]      ~ all                  = false (type:boolean)
[config]      ~ json                 = false (type:boolean)
[config]      ~ raw                  = false (type:boolean)
[config]      ~ softreset            = false (type:boolean)
[config]      ~ noninteractive       = false (type:boolean)
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: print("echo1337")
ERROR
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 
[NodeMCU-Tool]~ Error: Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ?
    at Timeout._onTimeout (/home/dev/.nvm/versions/node/v12.3.1/lib/node_modules/nodemcu-tool/lib/connector/check-connection.js:13:20)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7) 
```

-------------
### issue Description ###
```
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 
```

### Detailed Description ###
idk what was going on here
```
☁  research  nodemcu-tool run init_webserv.lua 
[config]      ~ Project based configuration loaded 
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ Timeout, no response detected - is NodeMCU online and the Lua interpreter ready ? 
```

AndiDittrich, 2020-10-12:
i would assume that it is caused by the following issue: https://github.com/AndiDittrich/NodeMCU-Tool/blob/master/docs/Reset_on_Connect.md

gpr-indevelopment, 2020-12-31:
I have the very same problem, and the solution on the documentation did not work for me... Even tried with 5000 delay.

gpr-indevelopment, 2020-12-31:
Actually, the command I was trying to run is not `fsinfo`. I was getting the error on `mkfs`.
It worked for me when running `fsinfo` before `mkfs` using the recommended connection delay. Thank you @AndiDittrich

weskerfoot, 2021-05-26:
I had the same issue, and some combination of these things helped fix it for me:

1) Hold down "flash" button, and "reset" button at the same time.
2) Release the "reset" button
3) Reflash firmware (while still holding the "flash" button)
4) Reset
5) Reflash again
6) Uploading works again.

I'm not sure what happened exactly, but this worked for me.

closed 2021-05-26 by AndiDittrich