NodeMCU-Tool/NodeMCU-Tool #80: fsinfo always detects 0 files, even if there are files on the device

quetzalcoatl, 2020-10-11:
## Checklist ##

1. YES, I can establish a serial connection both with Putty and `NodeMCU-Tool terminal`. 
2. Does the `nodemcu-tool fsinfo` command work? Answer: PARTIALLY, and that's what this issue is about.

-------------

## Environment ##

* **Operating System**  
  Windows 10-2004
* **ESP Device/Revision**  
  ESP8266
  looks very much like the Original NodeMCU `v2.0`, but probably a clone
* **NodeMCU-Tool Version**  
  `3.2.1`
* **Node.js Version**  
  `8.12.0`
* **NodeMCU LUA Firmware Version**  
```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
        branch: release
        commit: 64bbf006898109b936fcc09478cbae9d099885a8
        release: 3.0-master_20200910
        release DTS: 202009090323
        SSL: false
        build type: float
        LFS: 0x40000 bytes total capacity
        modules: adc,bit,cron,encoder,file,gpio,gpio_pulse,i2c,net,node,ow,pwm2,rtctime,sigma_delta,sntp,softuart,spi,tmr,wifi
 build 2020-10-10 21:38 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

**Please note that there's `uart` module is NOT included.**

-------------

## Debug Output ##

```
C:\Users\XYZ\Desktop\esp8266>nodemcu-tool --debug --io-debug fsinfo
[config]      ~ debug mode enabled
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger
[config]      ~ Project based configuration loaded
[config]      ~ baudrate             = 115200 (type:string)
[config]      ~ port                 = COM10 (type:string)
[config]      ~ connectionDelay      = 0 (type:number)
[config]      ~ minify               = true (type:boolean)
[config]      ~ compile              = true (type:boolean)
[config]      ~ keeppath             = false (type:boolean)
[config]      ~ remotename           = null (type:object)
[config]      ~ run                  = false (type:boolean)
[config]      ~ all                  = false (type:boolean)
[config]      ~ json                 = false (type:boolean)
[config]      ~ raw                  = false (type:boolean)
[config]      ~ softreset            = false (type:boolean)
[config]      ~ noninteractive       = false (type:boolean)
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: p
[serialport]  ~ RX: rint("echo1337")
[serialport]  ~ RX: echo1337
>
[serialport]  ~ TX: print(node.info("hw"));
[serialport]  ~ RX: p
[serialport]  ~ RX: rint(node.info("hw"));
[serialport]  ~ RX: t
[serialport]  ~ RX: able: 0x3fff0148
>
[serialport]  ~ TX: for k,v in pairs(node.info("hw")) do print(k,v) end;for k,v in pairs(node.info("sw_version")) do print(k,v) end;for k,v in pairs(node.info("build_config")) do print(k,v) end;
[serialport]  ~ RX: f
[serialport]  ~ RX: or k,v in pairs(no
[serialport]  ~ RX: de.info("hw")) do print
[serialport]  ~ RX: (k,v) end;for k,v in pa
[serialport]  ~ RX: irs(node.info("sw_versi
[serialport]  ~ RX: on")) do print(k,v) end
[serialport]  ~ RX: ;for k,v in
[serialport]  ~ RX: pairs(node.info("build_config")) do print(k,v
[serialport]  ~ RX: ) end;
[serialport]  ~ RX: f
[serialport]  ~ RX: lash_size   4096
chip_id
[serialport]  ~ RX: 4378623
flash_mode      0
[serialport]  ~ RX: flash_speed 40000000
[serialport]  ~ RX: flash_id    1458208
[serialport]  ~ RX: g
[serialport]  ~ RX: it_branch   release
[serialport]  ~ RX: git_release 3.0-master_2020
[serialport]  ~ RX: 0910
git_commit_id   64bbf00689810
[serialport]  ~ RX: 9b936fcc09478cbae9d0998
[serialport]  ~ RX: 85a8
node_v
[serialport]  ~ RX: ersion_minor        0
git_commit_dts  202009090323
[serialport]  ~ RX: node_version_revision       0
[serialport]  ~ RX: node_version_major  3
[serialport]  ~ RX: s
[serialport]  ~ RX: sl
[serialport]  ~ RX: f
[serialport]  ~ RX: alse
[serialport]  ~ RX: num
[serialport]  ~ RX: ber_type
[serialport]  ~ RX: float
[serialport]  ~ RX: modules
[serialport]  ~ RX: adc,bit,cro
[serialport]  ~ RX: n,encoder,file,gpio,gp
[serialport]  ~ RX: io_pulse,i2c
[serialport]  ~ RX: ,net,node,ow
[serialport]  ~ RX: ,pwm2,rtcti
[serialport]  ~ RX: me,sigma_del
[serialport]  ~ RX: ta,sntp,sof
[serialport]  ~ RX: tuart,spi,tm
[serialport]  ~ RX: r,wifi
lfs
[serialport]  ~ RX: _size       262144
>
[NodeMCU-Tool]~ Connected
[device]      ~ Arch: esp8266 | Version: 3.0.0 | ChipID: 0x42cfff | FlashID: 0x164020
[serialport]  ~ TX: print(file.fsinfo())
[serialport]  ~ RX: p
[serialport]  ~ RX: rint(
[serialport]  ~ RX: file.fsinfo())
[serialport]  ~ RX: 3
[serialport]  ~ RX: 232378      4769    3237147
>
[serialport]  ~ TX: local l = file.list();for k,v in pairs(l) do uart.write(0,k..":"..v..";") end print("")
[serialport]  ~ RX: l
[serialport]  ~ RX: ocal l = file.list(
[serialport]  ~ RX: );for k,v in pairs(l) d
[serialport]  ~ RX: o uart.write
[serialport]  ~ RX: (0,k..":"..v..";") end
[serialport]  ~ RX: print("")
[serialport]  ~ RX: L
[serialport]  ~ RX: ua error:   stdin:1: attempt to index global 'uart' (a nil v
[serialport]  ~ RX: alue)
stack traceback:
        stdin:1: in main chunk
        [C]: ?
        [C]: ?
>
[device]      ~ Free Disk Space: 3156 KB | Total: 3161 KB | 0 Files
[device]      ~ No Files found - have you created the file-system?
[NodeMCU-Tool]~ disconnecting
[serialport]  ~ RX:
```

-------------

## Issue Description ##
You've probably already noticed that error in the debug-output:

`[serialport]  ~ RX: ua error:   stdin:1: attempt to index global 'uart'`

It comes from `lib/lua/esp8266-commands.js`  
from `listFiles` on line 32

The issue is gone after changing that line from
`    listFiles: 'local l = file.list();for k,v in pairs(l) do uart.write(0,k..":"..v..";") end print("")',`
to
`    listFiles: 'local l = file.list();for k,v in pairs(l) do print(k..":"..v..";") end print("")',`

[image]

AndiDittrich, 2020-10-12:
the uart module is required for NodeMCU-Tool - the whole communication is based on messages terminated by `\n` - it would require a lot rework to change this (of course, `\0` would be even better..)

quetzalcoatl, 2020-10-12:
`\n` - alright. I guessed so, that's why what I did in my hotfix

`the uart module is required for NodeMCU-Tool` - That's what I thought when I saw that line! And imagine my surprise when I realized that I actually do not have `uart` module on the device, and NodeMCU-Tool works! I can `terminal` to it, I can `fsinfo` (with my hotfix), I can upload/compile/minimize/etc the files. I have not tried `download`'ing files, but I suppose that it uses the `transferReadHelper` so it probably won't work as it is now, but `transferWriteHelper` certainly seems to work, as all files I uploaded were absolutely fine.

It seems that NodeMCU-Tool actually doesn't really rely that much on uart module being present. I reviewed some LUA code in the NodeMCU-Tool and it's mostly `print()`'ing that works fine w/o `uart`. Of course my knowledge about NodeMCU-Tool is limited, but so far I noticed only that `fsinfo` (and probably `transferReadHelper`) issues.

I'll probably try to tackle `transferReadHelper` and switching between uart-and-non-uart modes too (similar how `encoder` is optionally used), but I'm pretty new to esp8266/lua/nodemcu/tool so .. one thing at time :)