NodeMCU-Tool/NodeMCU-Tool #81: No response detected - is NodeMCU online and the Lua interpreter ready ?

liurenqiu520, 2020-10-13:
## Checklist ##

1. Can you establish a serial connection **WITHOUT** Nodemcu-Tool - e.g. with a serial terminal like Putty ?  NO

2. Does the `nodemcu-tool fsinfo` command work ?  NO

-------------

## Environment ##

* **Operating System**  
  macOS 10.15.7
* **ESP Device/Revision**  
 ESP32-S2
* **NodeMCU-Tool Version**  
  `3.2.1`
* **Node.js Version**  
  `12.16`
* **NodeMCU LUA Firmware Version**  
  `dev-esp32` self compile

## Debug Output ##
➜  bros nodemcu-tool upload init.lua --debug --io-debug
[config]      ~ debug mode enabled 
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger 
[config]      ~ Project based configuration loaded 
[config]      ~ baudrate             = 115200 (type:string) 
[config]      ~ port                 = /dev/cu.usbserial-0001 (type:string) 
[config]      ~ connectionDelay      = 0 (type:number) 
[config]      ~ minify               = false (type:boolean) 
[config]      ~ compile              = false (type:boolean) 
[config]      ~ keeppath             = false (type:boolean) 
[config]      ~ remotename           = null (type:object) 
[config]      ~ run                  = false (type:boolean) 
[config]      ~ all                  = false (type:boolean) 
[config]      ~ json                 = false (type:boolean) 
[config]      ~ raw                  = false (type:boolean) 
[config]      ~ softreset            = false (type:boolean) 
[config]      ~ noninteractive       = false (type:boolean) 
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: �;32mI (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Ty
[serialport]  ~ RX: pe ST Offset   Length[0[39m
[serialport]  ~ RX: m
I
[serialport]  ~ RX: (68) boot:
[serialport]  ~ RX: 0 nvs
[serialport]  ~ RX: WiF
[serialport]  ~ RX: i data
[serialport]  ~ RX: 01 02 000
[serialport]  ~ RX: 09000 00006
[serialport]  ~ RX: 000
[0[39m
[serialport]  ~ RX: ;32mI (75)
[serialport]  ~ RX: boot:  1 phy
[serialport]  ~ RX: _init
[serialport]  ~ RX: RF data
[serialport]  ~ RX: 01
[serialport]  ~ RX: 01 0000f000
[serialport]  ~ RX: 00001000[0[39m
[serialport]  ~ RX: m
I
[serialport]  ~ RX: (83) boot:
[serialport]  ~ RX: 2 factory
[serialport]  ~ RX: fact
[serialport]  ~ RX: ory app
[serialport]  ~ RX: 00 00 0001
[serialport]  ~ RX: 0000 001800
[serialport]  ~ RX: 00
[0;0[39m
[... 407 lines cut ...]
I (450) wifi:config NVS flash: enabled
I (450) wifi:config nano formating: disabled
I (450) wifi:Init dynamic tx buffer num: 32
I (460) wifi:Init data frame dynamic rx buffer num: 32
I (460) wifi:Init management frame dynamic rx buffer num: 32
I (470) wifi:Init management short buffer num: 32
I (470) wifi:Init static rx buffer size: 1600
I (480) wifi:Init static rx buffer num: 10
I (480) wifi:Init dynamic rx buffer num: 32

NodeMCU ESP32 built with Docker provided by frightanic.com
        branch: dev-esp32
        commit: 7c07f914836d73e22e862c149def8b3314b77e43
        SSL: true
        modules: -
 build created on 2020-10-11 02:58
 powered by Lua 5.1.4 on ESP-IDF -9e7082 on ESP-IDF -9e7082 on ESP-IDF -9e7082 on ESP-IDF -9e7082 on ESP-IDF -9e7082 on SDK IDF
I (610) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (620) wifi:mode : sta (40:f5:20:78:79:8c)
[NodeMCU-Tool]~ Unable to establish connection 
[NodeMCU-Tool]~ No response detected - is NodeMCU online and the Lua interpreter ready ? 
[NodeMCU-Tool]~ Error: No response detected - is NodeMCU online and the Lua interpreter ready ?
    at /usr/local/lib/node_modules/nodemcu-tool/lib/connector/check-connection.js:28:28 

## Issue Description ##

AndiDittrich, 2020-10-13:
please fix your initial issue (point 1) - if the connection is not working the issue is not related to NodeMCU Tool

liurenqiu520, 2020-10-13:
> please fix your initial issue (point 1) - if the connection is not working the issue is not related to NodeMCU Tool

YES, i can use ESPlorer connect

AndiDittrich, 2020-10-13:
did you read about the reset-on-connect issue ?

liurenqiu520, 2020-10-13:
thx, i got it

closed 2020-10-13 by liurenqiu520

HHHartmann, 2020-10-13:
Btw. ESP32-S2 is a different chip which currently is not supported by nodemcu firmware