NodeMCU-Tool/NodeMCU-Tool #95: download not creating expected file

dannica2k, 2023-04-14:
-------------

## Checklist ##

1. Can you establish a serial connection **WITHOUT** Nodemcu-Tool - e.g. with a serial terminal like Putty ?  
  **YES**  
  If not, the issue not related to NodeMCU-Tool - it might be a problem with your serial driver, NodeMCU hardware and/or firmware  

2. Does the `nodemcu-tool fsinfo` command work ?  
  **YES**

-------------

## Environment ##

* **Operating System**  
 Windows 10-1809
* **ESP Device/Revision**  
  ESP8266 
  Original NodeMCU `v3.1.0`
* **NodeMCU-Tool Version**  
  `3.1.0`
* **Node.js Version**  
  `16.14.0`
* **NodeMCU LUA Firmware Version**  
  `1.5.1`

-------------

## Debug Output ##

> **NOTE** In case you have issues with some commands of NodeMCU-Tool, please provide the output of the command includng the debug flags
>
> e.g. `nodemcu-tool --debug --io-debug info`

M:\>nodemcu-tool -p com3 --debug --io-debug download init.lua
[config]      ~ debug mode enabled
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger
(node:11412) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
[config]      ~ baudrate             = 115200 (type:string)
[config]      ~ port                 = com3 (type:string)
[config]      ~ connectionDelay      = 0 (type:number)
[config]      ~ minify               = false (type:boolean)
[config]      ~ compile              = false (type:boolean)
[config]      ~ keeppath             = false (type:boolean)
[config]      ~ remotename           = null (type:object)
[config]      ~ run                  = false (type:boolean)
[config]      ~ all                  = false (type:boolean)
[config]      ~ json                 = false (type:boolean)
[config]      ~ raw                  = false (type:boolean)
[config]      ~ softreset            = false (type:boolean)
[config]      ~ noninteractive       = false (type:boolean)
[NodeMCU-Tool]~ Local file "init.lua" already exist - new file renamed to "init.lua.1681430292205"
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: print("echo1337")
[serialport]  ~ RX: echo1337
[serialport]  ~ RX: >
[serialport]  ~ TX: print(node.info("hw"));
[serialport]  ~ RX: print(node.info("hw"));
[serialport]  ~ RX: table: 0x3fff0020
[serialport]  ~ RX: >
[serialport]  ~ TX: for k,v in pairs(node.info("hw")) do print(k,v) end;for k,v in pairs(node.info("sw_version")) do print(k,v) end;for k,v in pairs(node.info("build_config")) do print(k,v) end;
[serialport]  ~ RX: for k,v in pairs(node.info("hw")
[serialport]  ~ RX: ) do print(k,v) end;for k,v in p
[serialport]  ~ RX: airs(node.info("sw_version")) do
[serialport]  ~ RX: print(k,v) end;for k,v in pairs
[serialport]  ~ RX: (node.info("build_config")) do p
[serialport]  ~ RX: rint(k,v) end;
[serialport]  ~ RX: flash_size
[serialport]  ~ RX: 4096
[serialport]  ~ RX: chip_id
[serialport]  ~ RX: 2392542
[serialport]  ~ RX: flash_mode
[serialport]  ~ RX: 2
[serialport]  ~ RX: flash_speed
[serialport]  ~ RX: 40000000
[serialport]  ~ RX: flash_id
[serialport]  ~ RX: 1458208
[serialport]  ~ RX: git_branch
[serialport]  ~ RX: release
[serialport]  ~ RX: git_release
[serialport]  ~ RX:
[serialport]  ~ RX: git_commit_id
[serialport]  ~ RX: f25dc56d3c6213b8ac7ce46d1293466
[serialport]  ~ RX: 137746eae
node_version_minor
[serialport]  ~ RX: 0
[serialport]  ~ RX: git_commit_dts
[serialport]  ~ RX: 202112300746
[serialport]  ~ RX: node_version_revision       0
[serialport]  ~ RX: node_version_major
[serialport]  ~ RX: 3
[serialport]  ~ RX: ssl
[serialport]  ~ RX: false
[serialport]  ~ RX: number_type
[serialport]  ~ RX: float
[serialport]  ~ RX: modules
[serialport]  ~ RX: file,http,net,node,pwm2,sjson,t
[serialport]  ~ RX: mr,wifi
[serialport]  ~ RX: lfs_size
[serialport]  ~ RX: 0
[serialport]  ~ RX: >
[NodeMCU-Tool]~ Connected
[device]      ~ Arch: esp8266 | Version: 3.0.0 | ChipID: 0x2481de | FlashID: 0x164020
[NodeMCU-Tool]~ Downloading "init.lua" ...
[serialport]  ~ TX: function __nmtread()local b = encoder and encoder.toBase64 while true do c = file.read(b and 240 or 1) if c==nil then print('')break end uart.write(0, b and encoder.toBase64(c) or string.format('%02X', string.byte(c)))end print('') end
[serialport]  ~ RX: function __nmtread()local b = en
[serialport]  ~ RX: coder and encoder.toBase64 while
[serialport]  ~ RX: true do c = file.read(b and 240
[serialport]  ~ RX: or 1) if c==nil then print('')b
[serialport]  ~ RX: reak end uart.write(0, b and enc
[serialport]  ~ RX: oder.toBase64(c) or string.forma
[serialport]  ~ RX: t('%02X', string.byte(c)))end pr
[serialport]  ~ RX: int('') end
[serialport]  ~ RX: >
[serialport]  ~ TX: print(file.open("init.lua", "r"))
[serialport]  ~ RX: print(file.open("init.lua", "r")
[serialport]  ~ RX: )
[serialport]  ~ RX: userdata: 0x3fff0bd8
>
[serialport]  ~ TX: __nmtread()
[serialport]  ~ RX: __nmtread()
[serialport]  ~ RX: Lua error:
[serialport]  ~ RX: stdin:1: attempt to index globa
[serialport]  ~ RX: l 'uart' (a nil value)
stack traceback:
        stdin:1: in function
[serialport]  ~ RX: '__nmtread'
        stdin:1: in main c
[serialport]  ~ RX: hunk
        [C]: ?
        [C]: ?
>
[connector]   ~ Transfer-Encoding: base64
[serialport]  ~ TX: undefined
[serialport]  ~ RX: undefined
[serialport]  ~ RX: >>
[NodeMCU-Tool]~ Data Transfer complete!
[NodeMCU-Tool]~ File "init.lua.1681430292205" created
[NodeMCU-Tool]~ disconnecting

-------------

## Issue Description ##
I uploaded a test init.lua file, fully functional. I am looking to download it from 8266 on to my windows 10 computer
DOWNLOAD doesn't create the expected file, seems like it encounters an error on uart.

### Expected Behavior ###
init.lua created on local computer 

### Current Behavior ###
the file is created but the content is garbage:
.æž®º+²×bŸV­µé©¶Ú"×±‚Zj[š®Ö§Š[Ú–ç¬µ§$¶¶œy¶œ’Ë]Š}bû§rØ¨ŸÿçšÚÞiÛ-v)õŠyšŠw!ºy

### Steps to Reproduce ###

1. nodemcu-tool -p com3 upload init.lua
2. nodemcu-tool -p com3 download init.lua

### Detailed Description ###

...

### Possible Solution ###

HELP!

dannica2k, 2023-04-14:
One more observation, after the download fails, 8266 is stuck and needs to be reset:

M:\>nodemcu-tool -p com3 --debug --io-debug download init.lua
[config]      ~ debug mode enabled
[config]      ~ io-debug mode enabled - rx/tx messages are passed to logger
(node:29228) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
[config]      ~ baudrate             = 115200 (type:string)
[config]      ~ port                 = com3 (type:string)
[config]      ~ connectionDelay      = 0 (type:number)
[config]      ~ minify               = false (type:boolean)
[config]      ~ compile              = false (type:boolean)
[config]      ~ keeppath             = false (type:boolean)
[config]      ~ remotename           = null (type:object)
[config]      ~ run                  = false (type:boolean)
[config]      ~ all                  = false (type:boolean)
[config]      ~ json                 = false (type:boolean)
[config]      ~ raw                  = false (type:boolean)
[config]      ~ softreset            = false (type:boolean)
[config]      ~ noninteractive       = false (type:boolean)
[NodeMCU-Tool]~ Local file "init.lua" already exist - new file renamed to "init.lua.1681430887938"
[serialport]  ~ TX: print("echo1337")
[serialport]  ~ RX: print("echo1337")
[serialport]  ~ RX: Lua error:
[serialport]  ~ RX: stdin:2: '=' expected near 'pri
[serialport]  ~ RX: nt'
>
[connector]   ~ Echo: print("echo1337")
[connector]   ~ Response: Lua error:    stdin:2: '=' expected near 'print'
[NodeMCU-Tool]~ Unable to establish connection
[NodeMCU-Tool]~ No response detected - is NodeMCU online and the Lua interpreter ready ?
[NodeMCU-Tool]~ Error: No response detected - is NodeMCU online and the Lua interpreter ready ?
    at C:\Users\Daniel\AppData\Roaming\npm\node_modules\nodemcu-tool\lib\connector\check-connection.js:28:28

AndiDittrich, 2023-04-16:
the uart module is missing within your firmware