NodeMCU-Tool/NodeMCU-Tool #97: Resurrect this fantastic tool

marcelstoer, 2024-06-25:
This a plea for @AndiDittrich & @serg3295 to join forces and to bring NodeMCU-Tool back to the spotlight one way or the other.

- This project hasn't been updated in 3y.
- The fork at https://github.com/serg3295/NodeMCU-Tool/ does not seem to publish to the NPM repository. Instead it points to the original's npmjs.com package.
- The fork uses separate branches for up-to-date ESP32 support (e.g. auto and idf4).

I am sure the community would very much appreciate if the two of you could find a way to merge your efforts. Having a single NodeMCU-Tool project has lots of benefits for the community. Regardless of what comes out of this: thanks for what you've contributed so far!

serg3295, 2024-06-25:
Branches 'auto' and 'idf4' of the fork are outdated. The current branch is [dev-esp32](https://github.com/serg3295/NodeMCU-Tool/tree/dev-esp32).

I have no plan to publish my fork in NPM repository, as I am not the author of this application.
I can make a PR of the 'dev-esp32' branch.

marcelstoer, 2024-08-19:
> I can make a PR of the 'dev-esp32' branch.

I would certainly appreciate that. However, without commitment or interest from @AndiDittrich this would be in vain.

AndiDittrich, 2024-08-19:
dear @marcelstoer ,

currently i've absolutely no time to maintain/enhance this project

marcelstoer, 2024-08-19:
Np, I know how that feels. @serg3295 did a great job making your tool compatible with current versions of the ESP32 releases (keeping the ESP8266 support as-is). The question is, would you be willing to accept his PR and push a new release to the NPM repository i.e. minimal effort?

AndiDittrich, 2024-08-19:
"simple" is relative...merging requires some time...just took a quick look onto the branch

serg3295, 2024-08-21:
Hello, @AndiDittrich 

I opened a PR that contains only 3 commits to facilitate reviewing. These are the minimum changes to support esp32xx chips

In this PR:
- dependencies are not updated
- esbuild is not added
- warnings of eslint@8.57 not fixed

I can add commits with major upgrade for glob, commander and update dependencies if it will be necessary.

serg3295, 2024-08-22:
@AndiDittrich 

I updated all dependencies and opened two PRs.
First PR updates dependencies without breaking changes
Second one has a breaking change as `glob` requires `node >= 20`

You can simply drop these PRs if they don't make sense or break backward compatibility.

marcelstoer, 2024-08-22:
Thanks @serg3295, looks sensible. So, a pragmatic way forward could be to merge the first PR and release this as 3.x, and then merge the 2nd.

serg3295, 2024-08-22:
Perhaps. However, there is not much difference between node18 (for serialport) and node20 (for glob). It is a breaking change anyway.

AndiDittrich, 2024-08-22:
nodejs 18 ist required for current debian stable

serg3295, 2024-08-22:
I changed  `glob` version to 10.4.4 that runs on node18 and force pushed the second PR.

marcelstoer, 2024-12-22:
(somewhat related to #101)

@AndiDittrich allow me to take you up on your initial comment here

> currently i've absolutely no time to maintain/enhance this project

It's sad to hear, but I can very much relate to that. I know from personal experience how demanding it can be to maintain even small projects consistently over a long period of time. Sometimes it's the lack of extra free-time, sometimes the lack of motivation, sometimes the lack of community support, sometimes other personal stuff...and sometimes a bit of all of that.

Long story short, I would hate to see this great project become yet another slowly dying OSS project.

Would you be willing to share the maintenance burden? I could review PRs and help with cutting releases and pushing them to the NPM registry.

AndiDittrich, 2025-03-15:
@marcelstoer 

i just create the NodeMCU-Tool org and moved the repository. in case you're still interested in maintaining some parts of the source i would send you an invitation

marcelstoer, 2025-03-15:
Yes, still up for that. Maybe invite @serg3295 as well?

marcelstoer, 2025-03-16:
Thanks for your willingness to open up this repository by transferring into into an org.

closed 2025-03-16 by marcelstoer