NodeMCU-Tool/NodeMCU-Tool #103: Integrate CI again

marcelstoer, 2025-03-15:
There's currently no CI integration anymore but a defunct `.travis.yml` and an obsolete Travis CI badge in the README.

- Add a GH Actions workflow. Use @serg3295's  https://github.com/serg3295/NodeMCU-Tool/commit/43c181eb031089839fd429049b94a356eee0e941 for inspiration.
- Delete the Travis file.
- Replace the Travis badge in the README with a GH Workflow badge.

AndiDittrich, 2025-03-15:
would suggest to remove the package from npm and use the github repo url in the docs

marcelstoer, 2025-03-16:
I feel it be a real pity if we were to drop the NPM registry. Publishing is trivial with a GitHub Actions workflow. Once everything else is done I can set a workflow to which you'd only need to add an NPM deploy token (GitHub repository secret).

closed 2025-03-28 by marcelstoer

marcelstoer, 2025-03-29:
CI pipeline done, release pipeline pending.

commit baae4701 (2025-03-28): Add minimal CI workflow

Fixes #103
commit bc871ace (2025-03-28): Add minimal CI workflow

Fixes #103
commit 7e8dd7e2 (2025-03-28): Add minimal CI workflow

Fixes #103