NodeMCU-Tool/NodeMCU-Tool #105: Modernize the linting setup

marcelstoer, 2025-03-16:
With #99 @serg3295 upgraded eslint from 4.x to 8.x. While working on #103 I noticed two things:
- I can't currently even do a successful `yarn test` as there's a linting error.
- We should probably bump eslint to 9.x. This, however, requires to migrate the `.eslintrc.json` file.

@serg3295, I notice that you already migrated everything in your `dev-esp32`. Would you be willing to contribute a consolidated update?

serg3295, 2025-03-16:
Sure. I will prepare a PR soon.

marcelstoer, 2025-03-17:
Thank you so much 👍 

I don't mean to offload the whole maintenance effort to you. However, I noticed that you wen through a couple of iterations to get the linting setup right.

Maybe, just maybe, once we're done updating this upstream project, you won't even see the need to maintain a separate fork anymore.

serg3295, 2025-03-17:
Thank **you** for maintaining the project.
I will be happy to assist in the development of this project.

PR #106 (2025-03-23): Migrate to eslint v9
The first commit should solve the linting issue.
Migrating to eslint v9 is optional.

Fixes #105
closed 2025-03-23 by marcelstoer